svenhb/GRBL-Plotter #369: When saving gcode, non-gcode lines appear

lfmmd, 2024-01-02:
**Describe the bug**
When saving gcode, non-gcode lines appear

**To Reproduce**
Steps to reproduce the behavior:
1.run
2. file
3. openfile
4. select dxf file
5. Directly close the parameter prompt window that appears
6. ctrl + s  save .nc file

**Expected behavior**
Under normal circumstances, it should output clean gcode code without appearing like <Collection">
[file.zip](https://github.com/svenhb/GRBL-Plotter/files/13808010/file.zip)
The situation is different without actively closing the parameter prompt window.
I have included in the file what I think is necessary, if anything else is needed please let me know

lfmmd, 2024-01-02:
The problem occurred when I upgraded from 1.6.9.4 to 1.7.1.0

svenhb, 2024-01-02:
I assume you used this options?
[image]
[image]

I can't find a bug. Could you repoduce with 1.7.1.0?
Perhaps you modified the G-code manually before saving?

lfmmd, 2024-01-03:
[image]
I couldn't find it “remove all comments when saveing”   Can you provide the path for setting up

lfmmd, 2024-01-03:
[image]
I retested and found that after actively closing this window, the gcode still showed contamination

svenhb, 2024-01-03:
Can you upload the original DXF file and the resulting G-Code, without this option to remove comments?
[image]

lfmmd, 2024-01-05:
[dxfandfile.zip](https://github.com/svenhb/GRBL-Plotter/files/13838417/dxfandfile.zip)
I'm really sorry, the file I uploaded last time did not include dxf.

lfmmd, 2024-01-05:
> 您能否上传原始 DXF 文件和生成的 G 代码，而不使用此选项来删除注释？ ![图像](https://private-user-images.githubusercontent.com/19973251/293861760-6e22705b-2d36-49b7-926c-ca06aa83c41a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ0MzQ2MTEsIm5iZiI6MTcwNDQzNDMxMSwicGF0aCI6Ii8xOTk3MzI1MS8yOTM4NjE3NjAtNmUyMjcwNWItMmQzNi00OWI3LTkyNmMtY2EwNmFhODNjNDFhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA1VDA1NTgzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU3ZjMxZWM1M2E0MmYzNzdkNDFlNjliYzQwOWNiN2U1N2JmOTFlYmEwNDQ1MjkyNzI5ZGY5NmI1NDFkNzVlZGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.dH-E7mvu6JuUCTzG78I--ZPDuTOY0dF2DpAZxFEDR0c)

I didn't find this option on the software

svenhb, 2024-01-05:
[image]

svenhb, 2024-01-05:
Also if I import your DXF file and save it with "Remove all commands" and load the saved gcode file, there is no problem.

closed 2024-01-11 by lfmmd