svenhb/GRBL-Plotter #370: Question/Help : Tool diameter from tool table

JoeBeef, 2024-01-07:
Hi,

I am importing an SVG file into GRBL-Plotter to send to my CNC. The setup is made to take the information from the tool table.
I was expecting to see the tool diameter to be taken into account when generating the Gcode but this is not the case.

Am I missing a setup some where to tell to offset the toolpath of half the tool diameter ? 

Thanks in advance for your help.

svenhb, 2024-01-07:
You are right, I never paid attention to it, beacause it doesn't affect the generated G-Code - just 2D-view.
Will try to fix it.

JoeBeef, 2024-01-07:
Thanks for looking into it.

But at the sametime it brings more question.

When you say that just the 2D view is affected... I don't even see the 2D view changing.

Looking at the generated GCODE, I noticed the PenWidth = 0.004 parameter. Changing this value in the GCODE does affect the 2D view,
Now, my question is , where does it get that value from, I have looked everywhere for it an never found it ?

_( SVG import by GRBL-Plotter 1.7.1.0.c )
( Source: C:\Users\Maiso\Downloads\Dixies.svg )
(<Header >)
( G-Code lines: 5700 )
( Pen Down/Up : 37 times )
( Duration ca.: 2.6 min. )
( Conv. time  : 00:00:00.0526929 )
( Values from tool-table: spindle speed. XY feed. Z Values )
( Tool changes: 1)
( 1 ToolNr: 01 Cnt:   1  Name: Red=[000000] )
( SVG dimension : 39.00 61.00 )
( Title: svgOutput)
(</Header >)
F1000 (Setup - GCode-Header)
G90
G00 Z2.000
(<Figure Id="1" Geometry="path" PenColor="000000" ****PenWidth="0.004**"** PathLength="197.8"> )
M05 (Stop spindle - Option Z-Axis)
T01 M06 (Tool:1 Red=[000000])
M3 S1000 (Start spindle - Option Z-Axis)
G04 P1 
...._

Thanks,

JoeBeef, 2024-01-07:
oops Closed by mistake...

svenhb, 2024-01-07:
The pen-width is read from the imported graphics.
If you import SVG, the key-word is stroke-width (usally given in pixel) which will be converted to mm.

svenhb, 2024-01-07:
Example SVG with stroke-width of 0.5, 1, 2, 3.
You can drag and drop the picture below into GRBL-Plotter (on 2D view)
[image]

JoeBeef, 2024-01-07:
Thanks, just found out that the penValue and color comes from the SVG file!!!
I am new at this and I should have done more research.

Thanks,

closed 2024-01-07 by JoeBeef