svenhb/GRBL-Plotter #371: Hex file named as from GRBL V1.1 actually from GRBL V0.9i

SteveJgithub, 2024-01-07:
**Describe the bug**
**file v1.1_GRBL_coreXY_servo___.ino.standard.hex in release GRBL-Plotter-1.7.1.**0 seems to be based on v 0.9i of GRBL (not on v1.1h)  as, when uploaded to the Arduino using Xloader, Xloader logs loading 29790 bytes and  GRBL responds to the $I command with the GRBL 0.9i credentials. 

**To Reproduce**
use Xloader to upload that hex file, connect and issue a $I command from the console

**Expected behavior**
the response to the $I command should be something like "" [VER:1.1h.20190825:] [OPT:V,15,128]"

**Screenshots**
N/A

svenhb, 2024-01-07:
Yes, too bad. I took the hex file from here without testing: https://w.electrodragon.com/w/Arduino_CNC#Further_Use.2C_Documentation

And it is not version 1.1, because it is not accepting jog-command https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging

svenhb, 2024-01-07:
Here is a links to an other corexy version (probably without servo support); https://github.com/svenhb/GRBL-Plotter/issues/199#issuecomment-830804778

svenhb, 2024-01-07:
Check here for new hex versions of grbl v1.1h for Uno or Nano: https://github.com/svenhb/GRBL-Plotter/tree/master/Firmware/hex_files
Compiled, using following options:
Config.	Options
- I	COREXY
- J	COREXY and servo
- K	COREXY and servo and servo-off inverted

svenhb, 2024-01-24:
Now part of new setup: https://github.com/svenhb/GRBL-Plotter/releases

closed 2024-01-24 by svenhb