svenhb/GRBL-Plotter #372: Upload of  v1.1_GRBL_coreXY_servo___.ino.with_bootloader.standard.hex fails

SteveJgithub, 2024-01-07:
**Describe the bug**
An upload of v1.1_GRBL_coreXY_servo___.ino.with_bootloader.standard.hex to Arduino Uno using Xloader fails.

**To Reproduce**
Launch Xloader specifying the file name above
Wait for Xloader to run to completion
View error message

**Expected behavior**
Expecting a successful upload with a message giving the number of bytes loaded as per usual

**Screenshots**
N/A

Thanks

svenhb, 2024-01-07:
I will remove both files from the setup.
What setup do you need? CoreXY - with or without servo?

svenhb, 2024-01-24:
Check new hex files: https://github.com/svenhb/GRBL-Plotter/tree/master/Firmware/hex_files
Also part of new setup: https://github.com/svenhb/GRBL-Plotter/releases

closed 2024-01-24 by svenhb