svenhb/GRBL-Plotter #373: The new version of the software (V1.7.1.1) does not display points in the dxf file

lianzaozi, 2024-01-25:
The software version v1.7.1.1 can correctly parse points in dxf files but will not display points

Plotter software display
[image]

CAD software display
[image]

Test dxf
[1.zip](https://github.com/svenhb/GRBL-Plotter/files/14049053/1.zip)

svenhb, 2024-01-25:
Yes, the problem is again your scaling: this drawing has an width of 479842 mm, the point marker has a size of 4 mm.
If you zoom in you find the point markers.
Any idea how to fix it? Set point marker-size to x% of complete graphics-width?

lianzaozi, 2024-01-25:
> 是的，问题又出在你的缩放上：该图的宽度为 479842 毫米，点标记的尺寸为 4 毫米。 如果放大，您会发现点标记。 知道如何修复它吗？将点标记大小设置为完整图形宽度的 x%？

I found that in CAD software, when zooming in and out on point entities, the size of the displayed points in CAD does not change with scaling and is a fixed value.Zooming in and out only changes the distance displayed between points (the display size of points does not change with the scaling ratio, and this feature only applies to point entities)

Screenshot of reduced point display
[image]

Screenshot of enlarged point display
[image]

lianzaozi, 2024-01-25:
Yes, as you described, I enlarged the graphics to a large size and indeed saw the plotter software displaying the points

lianzaozi, 2024-01-25:
Thank you for your hard work

lianzaozi, 2024-01-25:
The software version of the plotter used for testing is V1.7.1.1

I have a question that I don't understand. I would like to ask you a question. My understanding is that whether arcs and circles generate G01 or G02/G03 is determined by the checkbox "Average G2/3 (Arc Move) commands, draw arcs/circles with line segments". After testing, checking and unchecking "Average G2/3 (Arc Move) commands, draw arcs/circles with line segments", the G code for generating arcs and circles is the same

But after verification, I found that after checking "Process Dashed Line Pattern", arcs and circles will generate G01, while unchecking "Process Dashed Line Pattern" will generate G02/G03

After selecting "Process Dashed Line Pattern", the arc and circle will generate screenshots of G01
[image]

Uncheck the "Process Dashed Line Pattern" to generate screenshots of G02/G03 for arcs and circles
[image]

Checking and unchecking "Average G2/3 (Arc Move) commands, draw arcs/circles with line segments" The G code generated by arcs and circles is the same
[image]

Dxf file for testing

[1.zip](https://github.com/svenhb/GRBL-Plotter/files/14052996/1.zip)

svenhb, 2024-01-25:
>After selecting "Process Dashed Line Pattern",
There is no other way to draw a circle with dashed line pattern then using G01 commands.
G02/G03 will draw an arc or  circle with no interruption.

lianzaozi, 2024-01-25:
The circles used for testing are drawn using solid lines instead of dashed lines
[image]

svenhb, 2024-01-25:
If this option is enabled, all possible G02/G03 code will be converted to G01:
[image]

lianzaozi, 2024-01-25:
> "Average G2/3 (Arc Move) commands, draw arcs/circles with line segments"

What does the "Average G2/3 (Arc Move) commands, draw arcs/circles with line segments" check box do?

svenhb, 2024-01-25:
If you have no other options enabled:
You may want G02/G03 code for short program (scaling aspect ratio must be 1:1) or
you may want G01 for circles to be able to scale the graphics independed from fixed aspect ratio 1:1

svenhb, 2024-01-25:
Try update: https://github.com/svenhb/GRBL-Plotter/releases/download/v1.7.1.1/GRBL-Plotter_Setup_1711a.exe

lianzaozi, 2024-01-25:
> Try update: https://github.com/svenhb/GRBL-Plotter/releases/download/v1.7.1.1/GRBL-Plotter_Setup_1711a.exe

https://github.com/svenhb/GRBL-Plotter/issues/373#issuecomment-1910728367

closed 2024-03-16 by svenhb