svenhb/GRBL-Plotter #374: run under linux

megahacker91, 2024-01-25:
is there any way to run this app under linux ???

svenhb, 2024-01-25:
It may work the same was as for lasergrbl:
It is easy to install LaserGRBL in linux using PlayOnLinux via this script

As alternative you can run LaserGRBL in linux with wine 5.0 and wine mono 4.9.4 (wineprefix 32-bit) and windowsdll gdiplus. read more

svenhb, 2024-02-27:
Did you try this?
https://forum.winehq.org/viewtopic.php?t=32363

metzger100, 2024-02-27:
Thanks for the link.
There were some fonts missing. Installing https://aur.archlinux.org/packages/ttf-ms-win11-auto fixed the font problem.
After installing it wine freezed but it worked flawlessly with GE-Proton8-27 by adding GRBL-Plotter as a non-Steam Game and and enabeling Steam Play.

metzger100, 2024-02-27:
If you could tell me which fonts are needed by grbl plotter I could write a short guide for linux for your Readme.
The main problem is, that every distro has different packages to install windows fonts and not always contain the same fonts.

svenhb, 2024-02-27:
I try to figure out, usually system fonts
Your error code probably pointed to this code line in virtual joystick code : 
AutoScaleMode = System.Windows.Forms.AutoScaleMode.Font

svenhb, 2024-02-27:
Default font for the controls is 
"Microsoft Sans Serif"

metzger100, 2024-02-27:
Great. I will open a pull request, when I think everything works. I didn't use grbl plotter to control my cnc yet. I first will test some stuff to verify everything works.

metzger100, 2024-02-28:
[GRBL-Plotter-Linux.md](https://github.com/svenhb/GRBL-Plotter/files/14433077/GRBL-Plotter-Linux.md)

I can't garantee that it will work on all distros but I tried it on a fresh Arch Installation and there it worked.f

closed 2024-12-15 by svenhb