svenhb/GRBL-Plotter #375: Version V1.7.1.1.a: The plotter software encountered an error while reading entities stored in blocks with inconsistent scaling ratios in the X and Y directions

lianzaozi, 2024-02-01:
Version V1.7.1.1.a: The plotter software encountered an error while reading entities stored in blocks with inconsistent scaling ratios in the X and Y directions.

The different scaling ratios in the X and Y directions of the entities in the block in the dfx file can cause errors in the plotter software's reading.

The software version of the plotter used for testing is v1.7.1.1.a, and I guess this issue should exist in all versions.

[image]
Screenshot of CAD software

[image]
Screenshot of the plotter software

The dxf file used for testing
[1 - (2).zip](https://github.com/svenhb/GRBL-Plotter/files/14122236/1.-.2.zip)

The color display of graphics in the drawing software when reading dxf is different from what is displayed in the CAD software

[image]

svenhb, 2024-02-03:
Thanks again for finding a bug 😉
I found the problem, will release a new version soon.

svenhb, 2024-02-04:
Check latest release: https://github.com/svenhb/GRBL-Plotter/releases

lianzaozi, 2024-02-04:
Thank you very much for your hard work. There are still some issues with version v1.7.1.2, and the areas indicated by A, B, and C circled in the screenshot still have issues.

[image]

The polyline and point at point A are saved in the same block. I guess the problem at point A is related to rotation

The arc and line at point B are saved in the same block.

The circle and line at point C are saved in the same block.

There is still an issue with parsing circles, arcs, and polylines in the block.

There is still a problem with the resolution of graphic colors. The colors displayed on the plotter are different from those in CAD.

[image]

svenhb, 2024-02-05:
My reference is LibreCAD, there your drawing looks like this, which is close to the GRBL-Plotter output:
Perhaps there are incompatibility problems with AutoCAD versions? LibreCAD can save up to "DXF 2007".
[image]

lianzaozi, 2024-02-05:
I am using AutoCAD, and the version saved in DXF is 2018

svenhb, 2024-02-05:
Can you save with an older DXF version?
I can't support the latest AutoCAD feature because there's no information to be found on the internet - I just found DXF 2012.
To be fair, you should use freeware tools to test GRBL-Plotter.
BTW. also Inkscape can't read your DXF file....

lianzaozi, 2024-02-05:
Latest dxf format reference document

https://help.autodesk.com/view/OARX/2024/ENU/?guid=GUID-A85E8E67-27CD-4C59-BE61-4DC9FADBE74A

svenhb, 2024-02-25:
Try new release: https://github.com/svenhb/GRBL-Plotter/releases

lianzaozi, 2024-02-26:
> Try new release: https://github.com/svenhb/GRBL-Plotter/releases

Thank you for updating the new version. When testing version 1.7.3.0 of the plotter software, I found some issues. In the dxf file, point 1 in the following image coincides with the endpoint 3 of the elliptical arc, while points 2 and 4 coincide; However, when using version 1.7.3.0 plotter to read dxf, point 1 does not coincide with endpoint 3 of the elliptical arc, and points 2 and 4 do not coincide
[image]

lianzaozi, 2024-03-02:
Version 1.7.3.0 of the plotter software can only generate G01 when reading circles stored in blocks. After unchecking "Process drawn line pattern", arcs and circles will not generate G02 and G03.

Should the user be allowed to choose whether to generate G01 or G02/G03 for arcs when the scaling ratios in the x and Y directions of the circles and arcs stored in the block are the same; When the scaling ratios of circles and arcs stored in the block are different in the x and Y directions, arcs can only generate G01.
[image]

svenhb, 2024-03-02:
Will change the code: 
If dashpattern is enabled, but pattern is empty, then use G02/G03, otherwise G01 to be able to apply the pattern.

lianzaozi, 2024-03-16:
感谢你发布新版本,我将进行一些测试
Thank you for releasing the new version. I will conduct some testing

lianzaozi, 2024-03-16:
Version V1.7.3.1 still does not solve the problem of parsing the coordinate points stored in the block after the elliptical arc is rotated and scaled in the X and Y directions. I drew two points at the starting and ending points of the elliptical arc. If I observe that the starting and ending points of the elliptical arc do not coincide with the point in the plotter, it indicates that the coordinate values of the starting and ending points of the elliptical arc are parsed incorrectly. I did this to facilitate the observation of whether the calculated coordinate values are incorrect in the plotter after parsing.

At the same time, I will take screenshots of the coordinate values displayed in CAD and plotter for comparison of this dxf file

[image]
[image]
[image]
[image]
[image]

test dxf file
[1 - (2).zip](https://github.com/svenhb/GRBL-Plotter/files/14625483/1.-.2.zip)

lianzaozi, 2024-03-17:
Version 1.7.3.1 of the software does not have settings that allow users to decide whether circles and arcs should generate g02/g03 or g01. Currently, this version can only generate g02/g03 for circles.

svenhb, 2024-03-18:
>Version 1.7.3.1 of the software does not have settings that allow users to decide whether circles and arcs should generate g02/g03

No need for an extra setting: if (scaling.X == scaling.Y) G02/G03 will be used (if enabled).

lianzaozi, 2024-03-18:
Where is (if enabled) enabled?

svenhb, 2024-03-18:
>Where is (if enabled) enabled? 
If scaling X=scaling Y, what should I do to make the circle generate G02?

This checkbox must not be checked:
[image]

>If scaling X=scaling Y, what should I do to make the circle generate G01?

Then check "Avoid G2/3"

lianzaozi, 2024-03-18:
What is the difference between checking and not checking "Process Dashed Line Pattern"? Will this checkbox affect circle generation G01/G02/G03?

[image]

svenhb, 2024-03-18:
If you have a dashed line and want to plot it you need to check this option, then, G02/G03 can't be used, because of the interruptions in the drawing path. Otherwise a normal line will be drawn, even if the property is "dashed".
[image]

svenhb, 2024-03-19:
Yes

lianzaozi, 2024-03-19:
Thank you for your reply. After conducting some tests, the following are the test results, and one of them is incorrect

[image]

svenhb, 2024-03-19:
Then perhaps an other option was also enabled?
[image]

lianzaozi, 2024-03-19:
I haven't changed any other options. I'll upload the file I used for testing and give it a try
[test.zip](https://github.com/svenhb/GRBL-Plotter/files/14652688/test.zip)
[image]

svenhb, 2024-03-19:
Ok, there's a bug with DXF import: once the line pattern was set, it couldn't be reset.

svenhb, 2024-03-20:
About dash pattern: Try test version https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1731a.exe

The other problem (start / end pos of ellipse) I can't fix for now. Must be a rounding error, when converting angles back an forth.

lianzaozi, 2024-03-20:
> About dash pattern: Try test version https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1731a.exe
> 
> The other problem (start / end pos of ellipse) I can't fix for now. Must be a rounding error, when converting angles back an forth.

Thank you for your hard work. I will conduct some tests.

After my testing, if the elliptical arc does not exist in the block, the starting and ending coordinates parsed by the plotter software are correct. However, if the elliptical arc exists in the block, and the block is rotated and scaled, the plotter software will read the starting and ending coordinates parsed by the block again, which will be incorrect.

lianzaozi, 2024-03-21:
[image]

Elliptical arcs stored in blocks, incorrect resolution of starting and ending coordinates
[image]

It is not an elliptical arc stored in a block, and the coordinates of the starting and ending points are parsed incorrectly
[image]

test file
[1 - (2) -.zip](https://github.com/svenhb/GRBL-Plotter/files/14695161/1.-.2.-.zip)

lianzaozi, 2024-04-01:
[image]
[image]

The starting angle of an elliptical arc
(45/180)*3.1415926=0.785398163≠ 0.79741212786 (a2)

Termination angle of elliptical arc
(342.097334/180)*3.1415926=5.97072484≠ 5.0571035 (a1)

Perhaps the insufficient accuracy of the numerical values read in dxflib resulted in inaccurate angle values for the starting and ending points of the elliptical arc

test dxf
[1 - (4).zip](https://github.com/svenhb/GRBL-Plotter/files/14818698/1.-.4.zip)

lianzaozi, 2024-04-01:
[image]
I tried to draw more points on the elliptical arc and found that the coordinate values resolved by the entire elliptical arc are not accurate. In dxf, these points are completely on the elliptical arc. I will spend some time trying to solve this problem

svenhb, 2024-04-01:
Yes, as soon as there are different scalings for X and Y inside the block, the angles a1, a2 are no longer correct. 
If acalings are the same, then there is no problem.
Could not figure out why, up to now.

lianzaozi, 2024-04-01:
How to output these logs

[image]

My settings are as follows, but during debugging, there was no output of print logs
[image]

svenhb, 2024-04-01:
You may enable extend logging, by clicking into the marked area
[image]

lianzaozi, 2024-04-01:
I clicked on the part you selected in the red box, do I need to set any other options?

lianzaozi, 2024-04-01:
Hello, excuse me, I need your help
What does the variable diff represent below?
What does the condition for determining if here mean?
[image]

What does the variable a represent below???
[image]

svenhb, 2024-04-01:
the "large" variable is needed to determine the arc behavior: 
a is the axis-rotation 
https://www.w3.org/TR/SVG2/paths.html#PathDataEllipticalArcCommands

lianzaozi, 2024-04-03:
It seems that solving ellipses in the plotter code is not using elliptic parameter equations

I usually use the following equation to calculate elliptical trajectories
[image]
[image]

lianzaozi, 2024-04-03:
From what I understand the variable a in the program should compute t1;
But the actual variable a in plotter is equal to t2

I don't really understand why I'm calculating Angle t2, right? Can you help me explain?

[image]

[image]

lianzaozi, 2024-04-03:
I finally found the reason, not because the scaling ratio of x direction and Y direction was inconsistent, but because the elliptic arc was stored in the block with an Angle, which was 13.178191. The elliptic arc in plotter's parsing block did not take into account that the elliptic arc in the block also had an Angle. Now I have adjusted the Angle to 0, and plotter will The parsing is normal. So we should take this into account when analyzing the elliptic arc in the block

The elliptic arc in the block has no Angle screenshot
[The elliptic arc in the block has no Angle.zip](https://github.com/svenhb/GRBL-Plotter/files/14849607/The.elliptic.arc.in.the.block.has.no.Angle.zip)

[image]

Angular screenshot of the elliptic arc in the block
[The elliptic arc in the block has an Angle.zip](https://github.com/svenhb/GRBL-Plotter/files/14849611/The.elliptic.arc.in.the.block.has.an.Angle.zip)

[image]

Open the Elliptic Arc Block editor to measure the Angle of the screenshot
[image]

lianzaozi, 2024-04-03:
The x and y scales of the above two screenshot blocks are both 2:1

svenhb, 2024-04-03:
>but because the elliptic arc was stored in the block with an Angle, which was 13.178191

This angle is given by ellipse.MainAxis coordinates. (Endpoint of major axis, relative to the center)

lianzaozi, 2024-04-03:
YES,Does the current code include 13.178191 when calculating coordinates?

svenhb, 2024-04-03:
>Does the current code include 13.178191 when calculating coordinates?

Yes: Coordinates are Ellipse center: 407.181; 143.770  Mainaxis  -386.528; 90.504
-386.528; 90.504 results to an angle of -13.17
[image]

lianzaozi, 2024-04-04:
Hello, where are these logs stored?

[image]

svenhb, 2024-04-04:
Go to the Data Path/logfiles:
[image]

lianzaozi, 2024-04-04:
I know why the problem happened,(double w = ellipse.AxisRatio * (double)scaling.Y/(double)scaling.X;) By default, the major axis of the ellipse is parallel to the X-axis of the coordinate system, and the minor axis of the ellipse is parallel to the Y-axis of the coordinate system, but if the ellipse in the block has an Angle, the value of the variable w is not correct, and I think it should be calculated by affine transformation company.

[image]

lianzaozi, 2024-04-05:
.NET open source library for matrix operations, which I thought I might use when parsing ellipses

https://github.com/MKL-NET/MKL.NET

lianzaozi, 2024-04-12:
I have pinpointed the direction of the problem, or did I encounter a problem while parsing the scaling of the elliptical block? Period
If I set the scaling in the x and y directions separately, there will be problems.
The problem caused by scaling in the x direction is greater

[image]
test dxf(sx=0.16 sy=1 theta=0)
[sx=0.16.zip](https://github.com/svenhb/GRBL-Plotter/files/14962934/sx.0.16.zip)

[image]
test dxf(sx=1 sy=0.08 theta=0)
[sy=0.08.zip](https://github.com/svenhb/GRBL-Plotter/files/14962940/sy.0.08.zip)

lianzaozi, 2024-04-26:
I have identified the real reason for the incorrect parsing of ellipses and arcs within a block. The different scaling ratios in the X and Y directions are not the cause of this problem. The real reason is that if there are angles between the elliptical arcs and arcs within the block (the major axis of the ellipse is not parallel to the X direction of the coordinate system), this problem will occur

The difference between Ellipse 1 and Ellipse 2 in the ArSeg diagram is that Ellipse 1 has an angle within the block, while Ellipse 2 has no angle within the block; When inserting ellipses 1 and 2, the scaling ratios in the X and Y directions are different, but ellipses 1 are parsed correctly and ellipses 2 are parsed incorrectly
[image]

test dxf file
[123.zip](https://github.com/svenhb/GRBL-Plotter/files/15124561/123.zip)

svenhb, 2024-04-26:
I assume the current order is wrong: now 1st I apply xy scaling to the ellipse-properties, then I calculate the ellipse points. 
I will try to calculate ellipse points 1st, then apply xy scaling - but it is a huge work.

lianzaozi, 2024-04-26:
Firstly, obtain the coordinates before scaling, then use matrix operations to calculate the scaled coordinates, and then use the calculated coordinates to form a transformed ellipse. As I am not very familiar with programming, I can only provide you with the conclusion I have drawn after verification in CAD to provide you with an idea

svenhb, 2024-05-02:
I hope now it is correct, check https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1731g.exe

svenhb, 2024-05-03:
There is still a bug for circle with blockScaling.X != blockScaling.Y. But the fundamental problem has been found...

lianzaozi, 2024-05-06:
Thank you for updating the software. I will conduct some necessary tests.

svenhb, 2024-05-20:
Should be fixed now: https://github.com/svenhb/GRBL-Plotter/releases

closed 2024-05-20 by svenhb