svenhb/GRBL-Plotter #376: Process Automation Expansion

smartsoldier, 2024-02-05:
**Describe the solution you'd like**
Additional commands to be used in the Process Automation scripts that are able to call certain commands and menus inside the program. For example, more than just automating the load/stream of premade G-Code files or SVG, I could use the Process Automation tools to automate generation of G-Code items within the plotter (more like macros).
**Describe alternatives you've considered**
Currently, I am generating a simple QR-Code along with human readable text beside it, offsetting it, and streaming it to my laser. I do all of this within the plotter with the tools it provides. However, I must manually increment the QR-Code and the human readable lettering and regenerate each iteration.
**Additional context**
I am a novice VB user, so if the answer is to pull the source code and write my own scripts in Python or VB I completely understand, however I wouldn't be able to do that on my own.

svenhb, 2024-02-05:
I have to think about.

svenhb, 2024-02-06:
I think it is doable, but needs some time...
>Currently, I am generating a simple QR-Code along with human readable text beside it,

Can you explain more detailed (step by step) what this conains and what different data you have? And a screenshot of the result?
After processing one data-set you can't directly continue with the next dataset - correct? You need the change the workpice?
I assmue "data1" for creating text and "data2" as barcodedata? Will be provided via CSV textfile?

closed 2024-03-16 by svenhb