svenhb/GRBL-Plotter #378: Cannot connect via hc-05 bluetooth module

Srallie, 2024-02-18:
hello,

i have the mega x5 grbl firmware and connect them to a hc-05 bluetooth module. By other programs like laser grbl/ lightburn i will connect directly,but grbl plotter says:

* COM5 not available

*

how to solve?

svenhb, 2024-02-19:
Perhaps the bluetooth port has an other number?
Did you try to "Scan Ports" and find the correct port?
[image]
[image]

Srallie, 2024-02-19:
Hello,

Thanks for fast response. When i plug it in with cable, com 3 works perfect. When i turn on Bluetooth on com 5, then it didnt work.

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

image.png (view on web)<https://github.com/svenhb/GRBL-Plotter/assets/19973251/6eebf48c-196d-4836-aed5-169b19a6fa56>
image.png (view on web)<https://github.com/svenhb/GRBL-Plotter/assets/19973251/6a317987-2041-4668-a66f-5e177e0d554e>

svenhb, 2024-02-22:
I tried with my bare Mega 2560 (x5 grbl firmware) and it works...
[image]

Srallie, 2024-02-22:
Ok,strange..I will try again this week.👍

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

svenhb, 2024-02-23:
Perhaps thats the problem?
NOTE: powering the grbl-controller via PC-USB will cause problems, because RXD/TXD of PC-USB are connected to RXD/TXD of HC-05 module = short circuit of the data lines!

closed 2024-03-16 by svenhb