svenhb/GRBL-Plotter #379: Add Linux Installation Guide to the Wiki and the Website

metzger100, 2024-02-28:
Hey Sven,
I open a new Issue for that, because I think that the README.md is not the right place for this complex topic. I think it is better if it is in the wiki.
I wrote the guide for Linux. I didn't get it to run with wine but it worked with Proton (Experimental).
I really hope this is helpful for user of other distros as well because I only tested it on fresh Arch Installtion to verify it works. I tried to make the guide as generic as possible.

[GRBL-Plotter-Linux.md](https://github.com/svenhb/GRBL-Plotter/files/14433463/GRBL-Plotter-Linux.md)

I added also a Image to show how it has to look if it worked in `## Setup USB Port to connect to the CNC` at 5.

[image]

metzger100, 2024-02-28:
That Issue is related to #374

metzger100, 2024-02-28:
I refined the guide. There were some language errors and some incomplete informations in it. They are now fixed.
[GRBL-Plotter-Linux.md](https://github.com/svenhb/GRBL-Plotter/files/14439811/GRBL-Plotter-Linux.md)

svenhb, 2024-02-29:
Thanks alot, will do it at the weekend.

metzger100, 2024-03-06:
Thanks for commiting. Could you remove line 209? There is a wrong path in it.

svenhb, 2024-03-07:
ok, done.

closed 2024-03-07 by metzger100

metzger100, 2024-03-13:
I justed faced a crash using Proton Experimental which I could not reproduce on other Linux PCs. Switching to Proton GE fixed it somehow. You can easyly install it with ProtonUP-Qt or manually with the guide at: https://github.com/GloriousEggroll/proton-ge-custom