svenhb/GRBL-Plotter #380: Can I move genrated paths with different color together

lyg09270, 2024-03-02:
**Describe the solution you'd like"
I use a graphic to generate paths with different color.
When I trying move different path in same time.I find that I can only move the path one by one.This caused a litte bit offset in my drawing.Because I can only drag the paths one by one. 
[image]

Maybe I can select them together?I tried Ctrl-A and hold Ctrl to select more path.It dosen't work.

svenhb, 2024-03-02:
You could put the different paths into a same layer (e.g. in Inkscape) and then group by layer in GRBL-Plotter.
But anyway it would be nice to select different paths by shift-click as in other application. I will check if could implement this.

lyg09270, 2024-03-02:
I find that I can also move different path at once by this[image]
But i think shift-click is better.For instance,If I have two character in generated path.And I wanna move them closer."Apply offset" couldn't do that.since it move all path together.

svenhb, 2024-05-20:
Unfortunatly it is not possible to select different objects with the current software design. But I save now the last transform which can be applied to the next selected object: https://github.com/svenhb/GRBL-Plotter/releases

Do a transform on a selected form. select the next form, right click and select "Apply last transform"
[image]

commit 530be376 (2023-05-19): Testversion 1.6.9.5 19.05.2023

Try to fix #380 start of drag path
closed 2024-05-20 by svenhb