svenhb/GRBL-Plotter #382: Symbols used in 2D view

Marmadoc, 2024-03-10:
In 2D view I have 2 markers (ikon or symbols - i do not know).
One is black square - I understand it represents tool position,
but what is red square - is jumping when I click on 2D are - what it represents?
Maybe it is trivial question - but can't find answer.

svenhb, 2024-03-10:
The red marker shows the coordinate of the actual marked line:
- click into a specific G-Code line, the marker shows the coordinate in the 2-D view
- do alt-click in 2-D view to set marker to the closest code-coordinate, the coresponding code-line will be shown
- during streaming, the red marker shows the last sent coordinate, towards the grbl-controller

Marmadoc, 2024-03-10:
Thank you so much. Now it is easy.

closed 2024-03-16 by svenhb