svenhb/GRBL-Plotter #383: 2D line representation.

Marmadoc, 2024-03-10:
I'm very sorry but I have problem to find answer for my problem.
I have simple test g code file with three straight lines
But in 2D window preview it is only one color, one type of line - I can't recognize what is what
[image]
[image]
Can you point me how to fix it.
Thx in adavance

svenhb, 2024-03-11:
The Code was not generated by GRBL-Plotter. How to know where a line starts?
GRBL-Plotter uses G0 to move to next line and G1 to draw a line, it is also an indicator, when displaying the code. Also comments will be set for safety reasons: "(PU)" for pen up and "(PD)" for pen down...

The resulting code looks like:
```
F1000 (Setup - GCode-Header)
G90
M3 S10 (PU)
G04 P0.100
(<Figure Id="1" Geometry="Line" PenColor="000000" PenWidth="0.00" Layer="0" PathLength="36.1"> )
G00 X10.000 Y30.000  
M3 S80 (PD)
G04 P0.100
G01 X30.000 Y60.000 F1000 
M3 S10 (PU)
G04 P0.100
(</Figure>)
(<Figure Id="2" Geometry="Line" PenColor="000000" PenWidth="0.00" Layer="0" PathLength="31.6"> )
G00 X40.000 Y50.000  
M3 S80 (PD)
G04 P0.100
G01 X50.000 Y20.000 F1000 
M3 S10 (PU)
G04 P0.100
(</Figure>)
G0X0Y0 (Setup - GCode-Footer)
M30
```

[image]

Marmadoc, 2024-03-11:
Thank You very much. You are extremally helpful for me. I build my own cnc plotter and try to start using it.
Just one more question - if you do not mind ;
My plotter use Arduino Uno + CNC shield. In your opinion should I use standard GRBL firmware - or modified for pen plotter with servo motor - such as GRBL(i) firmware.
Once again -thank you very much

svenhb, 2024-03-11:
I would use the  modified, so you can directly control a RC-Servo (but don't forget the seperate power-source).
I have a selection of firmwares here (and also in the setup): https://github.com/svenhb/GRBL-Plotter/tree/master/Firmware/hex_files

Marmadoc, 2024-03-11:
Thank You very much. I'm afraid that my knowledge is not enough to play with hex files. I'm in stage that I have to have ino file to upload firmware to Arduino Uno. I will try to find the way. For now I have firmware for plotter.
Thx again

svenhb, 2024-03-11:
Just try via Xloader: https://grbl-plotter.de/index.php?id=quick-guide&setlang=en

Marmadoc, 2024-03-11:
I'll try - thank you

Marmadoc, 2024-03-12:
It is me again. I did try Xloader and now I have your GRBL hex version Grbl 1.1h. It works, however I have two questions:
1. M03 S10 - Pen Up PWM=10 Delay 0.10
M03 S1000 - Pen Down PWM=1000 Delay 0.10
Is any conversion from PWM to servo angle?
2. In Gcode before line M3 S1000 (PD) is line G01 Z-2.000 F1000 (PD) 
Is it correct or it is duplicated PD.

svenhb, 2024-03-12:
1. You need to check your settings for $30 and $31 it defines the usable range for the S value (https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#30---max-spindle-speed-rpm) inside grbl firmare it will be mapped to the range 7 - 38 which corresponds to the angle 0-180° (or 270° - depending on your servo) see https://grbl-plotter.de/index.php?id=quick-guide#pwm
2. if you get also Z coordinates, then you also enabled the Z axis which is mot needed:
[image]

svenhb, 2024-03-12:
So you could set $30 to 180 (if your max. servo turn is 180°), then the S value corresponds with the angle.

Marmadoc, 2024-03-12:
Thx - I forgot last time - G code concerted is for double path - where to change it?

svenhb, 2024-03-16:
I don't understand this question:
>G code concerted is for double path - where to change it?

Marmadoc, 2024-03-17:
[image]
As on screenshot.

Marmadoc, 2024-03-17:
Is it correct spelling?
[image]

closed 2024-04-16 by Marmadoc