svenhb/GRBL-Plotter #384: Tangential knife - reverse selected path bug

Kikenzi, 2024-03-14:
Hi there, 

if i want to try reverse selected path, so tool go originally from up to down and I want to reverse it - start down and go up:
[image]

All A axis positions go to 0.00 :
[image]

Is it a bug or I have missed some setting? 
Thank you for your help :)

svenhb, 2024-03-16:
Check latest release: https://github.com/svenhb/GRBL-Plotter/releases

svenhb, 2024-05-20:
Now included in new release: https://github.com/svenhb/GRBL-Plotter/releases

closed 2024-05-20 by svenhb