svenhb/GRBL-Plotter #385: Out of memory error with the drag knife path modification enabled?

vinevinz, 2024-03-18:
It always says the system is out of memory whenever I load new SVG with the drag knife compensation on in the setup. Whether it is drag knife or tangential knife produce same error. The file it self open and work just fine if path modification is off. ( I tried the file using pencil and it is working great)

Can you help point out what is wrong with the settings?

[image]

svenhb, 2024-03-18:
What kind of PC are you using? Operating system? RAM?
Can you upload the orginal SVG file?

vinevinz, 2024-03-18:
Step width circumfence settings totally fixed my issues! Thank you for the help. I am really new to this cnc things.

My PC is R5 3500, 16gb Ram, 3060. That is why I do not think spec is the issue.

closed 2024-03-18 by vinevinz

vinevinz, 2024-03-23:
Sorry for late reply,

My step width value was on default, I think it was 0.001 or 0.01.

Maybe showing warning about this when enabling Path Compensation would save beginner like me from this error.