svenhb/GRBL-Plotter #386: Homing Cycle and Home Button

Marmadoc, 2024-03-20:
1. Custom buttons - Homing 0;0;35 - what is purpose of this? It moves along X and Y (and Z) in increments 40, but what for it is?
2. Home Button - $H, here I have problem - after pressing Home button, servos clicks hold the torque but do not move. After some time Alarm is triggered.  Why? 
3. I like this software - but to find details on https://grbl-plotter.de/ - it is very difficult. Maybe search option can help?

svenhb, 2024-03-21:
1. It's a leftover from when I used my microplotter (without a limit switch). As a “homing” replacement, the unit was driven to the mechanical end points. http://svenhb.bplaced.net/?CNC___Mini_CNC_Plotter
2. Didi you enable homin in grbl? https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle
3. Thanks, will think about, did you see the alphabetic register (key-words)?  https://grbl-plotter.de/index.php?id=more&setlang=en

Marmadoc, 2024-03-21:
Yes - know about that - but it is not classic index. It is just links with few keywords.

Marmadoc, 2024-03-22:
I like to let you know that I switch back to GRBL(i) (not h hex file) - loaded via Arduino IDE - and now homing cycle is working (Home button) works perfectly.

closed 2024-04-16 by Marmadoc