svenhb/GRBL-Plotter #387: Loading gcode files and entering gcode manual

gpiotr1, 2024-03-22:
**Describe the bug**
I use the machine as a hobby and quite rarely. 
Last year (2023) with early versions of 1.7 I had an error where I couldn't load the gcode file into the program.
When I open a file. ngc, nothing happens in the code window (it is empty), the preview window is occupied (windows circle icon).
The same thing happens when I switch to code editing mode - the background turns yellow, I enter the code and I can't exit the editing mode (nothing happens).
Everything works OK on my other computers (windows 11, windows 10, Intel processors).
Komputerek, który steruje maszyną, to malutki stary HP t620 na AMD GX-415GA SOC. Windows 10Pro 22H2. Komputer służy tylko do obsługi maszyny. When I first encountered this problem, I downgraded grbl-plotter, but it didn't help. I reinstalled a clean Windows 10 system - the program started working properly even in the latest version. When Windows automatically deployed the updates, the problem returned.
The currently used version of the program is 1.7.3.1
I don't know at what point the program stopped working, after which update was installed (after installing a clean system from the image, there were several dozen of them)
grbl-plotter is the best program of this type and I don't want to change it :) Great job!

svenhb, 2024-03-22:
Do you see any error messages in the logfile?
[image]

gpiotr1, 2024-03-22:
The log is clean - no errors, but I noticed something else. When I open the file - the program does not open it. I click File->Setup, then the program 'thinks' for about 10 seconds. After this time, an options window opens and the previously loaded file opens in the background. Without clicking on options, I can wait for 10 minutes and nothing happens.
[logfile2.txt](https://github.com/svenhb/GRBL-Plotter/files/14722158/logfile2.txt)
[logfile3.txt](https://github.com/svenhb/GRBL-Plotter/files/14722159/logfile3.txt)
[logfile.txt](https://github.com/svenhb/GRBL-Plotter/files/14722160/logfile.txt)

svenhb, 2024-03-22:
Yes, not much to see, except the huge delay...
You may enable extend logging, by clicking into the marked area - just settings as shown should be enough...
[image]

gpiotr1, 2024-03-22:
Another observation - When I click on "edit mode" - the program switches to the "yellow" mode. I click "edit mode" again - the program hangs. I click on File->Options and after 10 seconds the program switches to working mode.
I loaded a simple gcode file - 10x10mm square. I ran the program - it executed once. When I tried to run it for the second time, the head moved to z=0 and the program hung.
[log_20240322.11.txt](https://github.com/svenhb/GRBL-Plotter/files/14722500/log_20240322.11.txt)
Generally, the File->Options menu opens for about 10 seconds each time. When I try to generate a code from an image, the program also does not respond, but when I enter the options, the code is displayed after 10 seconds.

gpiotr1, 2024-03-22:
Entering file->options 'refreshes' the program. I started drawing a 'square'. The program hung up while it was being executed (if I had the laser turned on, I would probably have a hole in the table). I clicked file->options and after 10 seconds the machine started and finished executing the code.
[logfile5.txt](https://github.com/svenhb/GRBL-Plotter/files/14722791/logfile5.txt)

Thank you very much for your interest in my problem and for your attention!

svenhb, 2024-03-22:
Can you try to run the setup and choose "Installation for me" - the 2nd option? Perhaps it is a windows-rights problem...

gpiotr1, 2024-04-03:
I'm sorry I haven't talked to you, but I was away from home for a few days.
Installation only for the user brought results - the application works and controls the machine.
It still hangs when trying to generate gcode from the system font (I tried it as a test). Options open in about 3-4 seconds.
I'm just learning how to generate code directly from the program. Previously, I only had a milling machine, now I have a second handle on which I attach a laser or a drag knife.
Once again, many thanks for your help!

closed 2024-04-03 by gpiotr1