svenhb/GRBL-Plotter #388: Tangential knife A-Axis turn command not getting included in applying height map

vinevinz, 2024-03-23:
**Describe the bug**

So, I upgraded the machine to tangential knife instead of drag. I'm using rotary cutter directly attached to A axis motor (works like charm).

Problem is, cutting mat is not that flat, probing height map results got me ~0.6mm difference from one edge to the other. As I'm applying the height map to G-Code, the A axis command actually getting removed.

Is this a bug or just the feature not implemented? Or maybe I am doing it wrong, bcs honestly I'm using 3d printer bed leveling logic here (it looks very similar anyway with klipper leveling).

I'm trying to cut circle (has not been successful yet due to uneven flatness issue)

G-Code screenshot attached

**To Reproduce**
Steps to reproduce the behavior:
1. Load circle SVG, tangential knife enable on A-Axis
2. Probe Z and height map (I am using 2 probe here, first for G38.2, second just limit switch for height map)
3. Height-map = Z-offset -1 (my probing tool offset), max value 0.0
4. Apply map
5. New g-code generated without A axis

**Expected behavior**
G-code includes A-Axis?

**Screenshots**
Original G-Code
[image]

Applied height map G-Code
[image]

Other example, A-axis turn command without lifting knife getting omitted when applying height map(this is untested code, please ignore for so much knife lifting)

[image]

Applied G-Code
[image]

Thank you!

svenhb, 2024-03-23:
I need to check

svenhb, 2024-03-23:
Try test version https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1731b.exe

vinevinz, 2024-03-24:
Sorry for late reply, kinda busy weekend here

It is working but not properly, I will attach the video of without height map and with the height map. Applying height map makes A axis goes kinda crazy on sharp corners and does not work on circles.

This is the normal WITHOUT HEIGHT MAP:

https://github.com/svenhb/GRBL-Plotter/assets/29923628/3d85cf84-4b4d-405c-b8fc-93107e422688

This 2 is with the height map:

https://github.com/svenhb/GRBL-Plotter/assets/29923628/1e6ec445-a9f0-421f-b070-3df4e12d6252

https://github.com/svenhb/GRBL-Plotter/assets/29923628/1664bd9d-5c0e-4128-9919-a5647e897246

Free SVG attached if you wanna try

[image]
[image]

vinevinz, 2024-03-24:
On my further analysis I think I found the problem, when applying height map to the g-code, the program doesn't respect the inserted turn command when re-writing it. It just copy whatever A-Axis command after Z lift pen.

The screenshot of the g-code after applying height map from this version of program 
[image]

Without height map applied, the program respects the turn command
[image]

svenhb, 2024-03-24:
Nice cutter 👍
Please try https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1731c.exe
Also arc conversion works now (tested with simple shape)

vinevinz, 2024-03-25:
Hi Sven, 

Thank you, I am building dual head cutter and pen with servo to lift up the pen.

My initial testing looks promising for this version. However circle only works by disabling G2/3 arc commands, is that okay? I am new to CNC world and I don't know if that will be a problem. Besides circle every other pattern works amazing. I think only moves translated from G2/3 commands is broken.

Enabling G2/3 will make A-axis spin uncontrollably which creates beautiful art on its own. Btw I load circle from SVG not simple shapes.
[image]

svenhb, 2024-03-25:
>However circle only works by disabling G2/3 arc commands,

Did you try allready version "c"? For me it works with G2/G3 command - also from SVG.
https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1731c.exe

vinevinz, 2024-03-30:
Hi sven, sorry for late reply it was crazy week here. 

Yes it is working perfectly, I think it was installation problem because I was lazy to do clean install. Thank you very much for your help on this problem.

Btw, I have new problem of start and ending edges not meeting. Can you help suggesting whether this is a software or a hardware problem?

[image]
[image]

svenhb, 2024-03-31:
I assume hardware problem: If you woud use a "regular knife" - perhaps the knife blade is too weak (too long) and bends sidewards, when cutting the circle - as a result the circle radius becomes bigger.

vinevinz, 2024-04-17:
Hi Sven sorry for late reply, been doing hard work this last 2 week

I have question regarding how does the software determine starting point?
Bcs I'm using rotary cutter which always overcut, I need to adjust the SVG so it will not overcut. It will be nice and easier to know where is the starting point when adjusting the design.

Example of starting point adjustment so, just make the starting point and end point separated by few millimeters.

[image]
[image]

Update on the machine:
I have redesigned everything for Z gantry and upgraded beefier X axis. You are right it is hardware problem. I'm feeling very silly to make it this strong just to cut 0.8mm synthetic leather, no wonder the leather cutting machine cost at least 10k usd 😅

[image]

svenhb, 2024-04-17:
If no sorting is active, the startpoint from the SVG will be used:
[image]

I have an option for overlap in closed figures (for laser cutting), Do you need to apply the "under cut" to each figure or just to open figures?
[image]

vinevinz, 2024-04-17:
Wow that is new info, I just knew SVG got a starting point! Thank you so much!

I think under cut in every figure is nice especially in closed path, I mean everytime the knife needs to lift up and change direction it needs to undercut.

svenhb, 2024-04-17:
I implemented path shortening, not sure if it meets your needs: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1731e.exe
[image]

vinevinz, 2024-04-20:
That one works great! Maybe just maybe if you could, can you add path shortening for the start point too?

Since rotary cutter will overcut the starting point and end point right. I will attached sample svg and picture about what I mean.

[image]

[image]

svenhb, 2024-04-20:
>add path shortening for the start point too

Yes, I can do. What should happen with paths which are shorten then the shortening? Delete at all?

svenhb, 2024-04-20:
Try this version: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1731f.exe

vinevinz, 2024-04-21:
Yes it works great now! Thank you so much sven for the hard work, you are writing 2 new features just for me. This is a very amazing feature to have.

> > add path shortening for the start point too
> 
> Yes, I can do. What should happen with paths which are shorten then the shortening? Delete at all?

I think on ideal world it will be nice to give out options to delete at all or ignore path shortening for that  path. But just delete at all will be fine. We can always fix it manually in the svg.

svenhb, 2024-04-21:
I'm glad to hear it's working. Feel free to give me a star.

vinevinz, 2024-05-07:
Hi Sven, I have completed my main build and I still have few questions for you, sorry for such an annoying user :(

1. Is there anyway to use tangential axis path modification only on specific tool? For example, I load and SVG that includes path for 2 tools, 1 knife and 1 pen, but I only want the path modification only work on knife. 
2. Is there anyway to move diffrent group path simultaneously? With the same example, I load SVG containing path for 2 tools, 1 pen and 1 knife, they will each create a group based on the layer sorting on the SVG. I can't seem to move it together bcos they need to work together on same object to achieve the desired result. One draw on top, one cut on the same piece, just like your laser and pen example.

Thanks before!

svenhb, 2024-05-07:
1) I experimented with machine settings given in SVG-Meta Data - perhaps thats a solution: https://grbl-plotter.de/index.php?id=form-setup-1#format-related
Write down whats missing to get it work.
2) Thats not possible yet, will try to implement it, but needs longer time....

vinevinz, 2024-05-08:
> 1. I experimented with machine settings given in SVG-Meta Data - perhaps thats a solution: https://grbl-plotter.de/index.php?id=form-setup-1#format-related
>    Write down whats missing to get it work.
> 2. Thats not possible yet, will try to implement it, but needs longer time....

1. Got it will try that out and get back to you.
2. That is fine, I was just asking if there's any hidden feature,

By the way, I found another bug!
Applying height map sometimes adding weird move to the A axis, which turns the blade 180 for a brief moment.

Take a look at line 434 and 435, the A axis doing 180 dance
[image]

[image]

Result and image attached

[image]

[image]

Surely this move instantly chip off the tip of the rotary blade ( well im using cheap chinese ones, gotta try with the olfa japan blades later)

svenhb, 2024-05-09:
From lines 433, 434 -> A0.51 to A-0.54 are just 1° difference, where is the 180° jump!? It is a motion almost only on +X (right) direction.
From the 2D view and your picture the motion must be almost only +Y or -Y (up or down) direction...
Can you upload the original graphic and show your import settings?

svenhb, 2024-05-09:
And also the height map

vinevinz, 2024-05-10:
> From lines 433, 434 -> A0.51 to A-0.54 are just 1° difference, where is the 180° jump!? It is a motion almost only on +X (right) direction. From the 2D view and your picture the motion must be [...]

Yes it is 1 degree difference but the blade goes from positive to negative value. Isn’t that will rotate the blade 180? I will share the file details later.

vinevinz, 2024-05-10:
Shared here:

1. SVG File
2. Generated G-Code with Height Map applied
3. Height Map
4. Machine settings

You could try the gcode with path simulation, it shows the blade direction turning 180

https://drive.google.com/drive/folders/1_cNrTTfut05U-jefynt6wD9UhelFIcta?usp=sharing

svenhb, 2024-05-10:
I could reproduce the problem with your data and fix. Please try: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1731h.exe

vinevinz, 2024-05-13:
> I could reproduce the problem with your data and fix. Please try: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1731h.exe
> 
> To reduce data, I would use this settings: ![image](https://private-user-images.githubusercontent.com/19973251/329523987-00b99e31-0973-49ae-9a66-44e0f67bd229.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU1OTk5ODAsIm5iZiI6MTcxNTU5OTY4MCwicGF0aCI6Ii8xOTk3MzI1MS8zMjk1MjM5ODctMDBiOTllMzEtMDk3My00OWFlLTlhNjYtNDRlMGY2N2JkMjI5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEzVDExMjgwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAyNzU0YzdmZTUxMGYwOTk0NTI1OTcyZjZkMmRlODUwNzllNTVkNDNjZWM5NTA2Njg2ZTcwNzQ3MmVhZTZjMjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.NKNX2mYiQ1w2MngRrsHKT-_62dqmQA0kq5V5VS9M4vI)

Hi sven,

this version works like charm fixing that bug. Thank you.

Apparently I can’t use your settings suggestion. Those settings will work fine with pen with bigger tip, but with blades only 0.3mm thick, it will cut jagged edges not smooth curves

svenhb, 2024-05-20:
Now included in new release: https://github.com/svenhb/GRBL-Plotter/releases

commit a6835c3d (2024-05-10): Test bug fix #388
closed 2024-05-20 by svenhb