svenhb/GRBL-Plotter #389: Set coordinate system in camera mounted on XY platform

azizikianoosh, 2024-03-27:
Thanks for what you have done so far.
I was checking the steps for camera setup in https://grbl-plotter.de/index.php?id=form-camera
and I found out setting offset in the version that I have is missing.
I wanted to know is there any reason that we do not have it or it is just missing.
II am using 1.7.3.0 revision. and here is my setup window for camera.
[image]
 I be grateful if you help me in this matter.

svenhb, 2024-03-27:
If you use "Camera mounted on XY platform" -> After doing the distortion corretion, there is no need for an offset, because center of video image points to the tool position.

azizikianoosh, 2024-03-27:
> If you use "Camera mounted on XY platform" -> After doing the distortion corretion, there is no need for an offset, because center of video image points to the tool position.

I want to mount it parallel to my tool and use it to find fiducials. I do not want to mount it in an angular position like the picture that I have sent.

svenhb, 2024-03-28:
Then you need to use "Camera mount on Z axis" but use same values for upper and lower position.

closed 2024-05-20 by svenhb