svenhb/GRBL-Plotter #390: Plot information

Marmadoc, 2024-04-12:
Hi.
As a plotter I have > 10 000 lines/figures. Software simplified it to 10 000 figures.
My question - can I get total path length?

svenhb, 2024-04-12:
You may increase this value:
[image]

Marmadoc, 2024-04-12:
I know about that. But how to get total length of all paths together? I saw on internet that somebody in description added - total length and total time. I do not know by what software. Maybe it is available in this software as well.

svenhb, 2024-04-12:
If you activate group, you can get the path length of the group:
[image]

svenhb, 2024-04-12:
I will check if I can add the total path length into the header section

Marmadoc, 2024-04-12:
Thank you for your hard work. Sounds fantastic. Maybe not header section but to G-Code dimension Window info.

svenhb, 2024-04-13:
Check new version: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1731d.exe

[image]

Marmadoc, 2024-04-14:
It is fantastic - thank you very much.
Only one more think - before I created my ini file (case). After new installation my file was deleted. Can you keep setup file during new installation?

svenhb, 2024-04-14:
Hm, deleting new files should not happen. If the file is from installation, but newer, you will be ask if it can be overwritten. Completly new files (created by you with uniqe name) should not be touched at all...

Marmadoc, 2024-04-14:
Perhaps i did something wrong. Before I install software in my own location.
Now installation is available only for default location "C".
My ini file is still in previous location, but I'm running out of space on drive C.

One more - maybe you didn't see my post before Graphic Imports -> GCode Generation ->
Lasemode - use S0 instead of M5 to switch off Laser
I think Lasemode is wrong spelling -> Lasermode.

svenhb, 2024-04-14:
With magnifier, the "r" is there...
[image]
But I separate the word for better reading:
[image]

Marmadoc, 2024-04-16:
super

Marmadoc, 2024-04-24:
Hi Svenhb. I do not know what I did but Path length info disappears from G Code dimension window. Is not in header either.
I did try to download this software from link what you gave me before - but is  not there - Page not Found.
Any idea how to fix it?

svenhb, 2024-04-24:
I updated the test release: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1731f.exe
Unreleased test version is placed on main page: https://github.com/svenhb/GRBL-Plotter

Marmadoc, 2024-05-05:
.

svenhb, 2024-05-20:
Now included in new release: https://github.com/svenhb/GRBL-Plotter/releases

closed 2024-05-20 by svenhb