svenhb/GRBL-Plotter #391: help plotting multi layer svg. manual pen change.

xn-78j0enb, 2024-04-15:
I want to plot a multi layer svg. How do I plot one layer at a time then return to 0,0 and pause for manual pen change?

I enabled these settings:
Graphics Import→G-Code Generation→General Options→Add Tool Change command Tx M06 , Add Pause (MO)
Graphics Import→Path Import→Grouping and Tools→Group objects of same property, group by layer.

when I run the simulation it correctly draws the first layer then continues on to the next layer without pausing.

svenhb, 2024-04-15:
The simulation does not take tool changes into account - sorry. You may check with your plotter and motor-power switched off.
Tool change option should look like this:
[image]

Marmadoc, 2024-04-15:
It is very important information for me as well.

svenhb, 2024-04-15:
To move to 0;0 for pen change, you need to set the move command here, like this:
[image]

xn-78j0enb, 2024-04-16:
This resolved my issue. Thank you.|
Great software. Great support.

closed 2024-04-16 by xn-78j0enb