svenhb/GRBL-Plotter #392: servo for pen up and pen down

Marmadoc, 2024-04-16:
My plotter with Your software is working.
I have only one problem. Pen down - it is very fast movement, and when hit the surface creates bigger dot.
I saw one your plotter with stepper motor for pen lifter. It works very well, what i Can see, because the movement is much slower.
I was thinking to switch from servo to stepper.
I can add extra arduino board to change servo speed, but
maybe it is possible  to add it to grbl software and add option speed for pen Up and pen Down.
I'm not a programmer but I know how to do this with extra arduino board.

svenhb, 2024-04-16:
You could try to work with a ramp.
1st find the zero value, where the pen just touches the paper:
[image]
2nd enable the ramp:
[image]

Then a line will be drawn by such code:
[image]

Marmadoc, 2024-04-16:
I do not have P91 Pen Zero

Marmadoc, 2024-04-16:
Now I'm stuck. I have Unknown Status - and can't do anything. Laser Grbl and UGS works but not GRBL Plotter.
Reinstall, restart, reload - nothing.
Any Idea what is wrong.

svenhb, 2024-04-17:
>I do not have P91 Pen Zero

Just press "Advanced settings"

>I have Unknown Status -

Are you sure the serial connection is done? Perhaps wrong port?
You may "Scan Ports" and try an other port?
[image]

Some time a hard reset is needed (my experience with Arduino mega) to be able to connect.

Marmadoc, 2024-04-17:
Thank You. Regarding servo - I have to try it.
Regarding U known Status - today all is OK. I do not know what was wrong before.
Thank you

Marmadoc, 2024-04-17:
A like to assign one button to Pen in Zero position.
For Lower Position the line is M3 S#GMIS
For Upper Position the line is M3 S#GMAS
What line to be for zero(middle) position M3 S#G???

Marmadoc, 2024-04-17:
Perfect. Thank you

closed 2024-04-17 by Marmadoc