svenhb/GRBL-Plotter #393: GRBL plotter with laser

Marmadoc, 2024-04-19:
I read that GRBLPlotter can be used with laser as well.
I like to try it.
I have LT-80W-AA PRO laser by LaserTree.
I have laser module - 3 wires, and laser module driver adapter P-DA-01 with two wires + Power 12V.
It looks like two wires are Signal and GND.
My question is can I use it with CNC shield V3 - like with servo Z+?
I understand I have to change GRBL setup - but will be OK with GRBLPlotter (instead of servo)?
Thx

svenhb, 2024-04-19:
Yes you need the "regular" grbl version which creates PWM from 0% to 100% - https://github.com/gnea/grbl
If you use the servo-grbl version your laser will be very week. Servo uses 1ms - 2ms high / 20 ms low, which results to an energy of 5% to 10%.

svenhb, 2024-04-19:
Here I combined plotter and laser: https://youtu.be/Ebe2kFlE058?si=TBMYBpiT18W6Bk85
But pen is controlled by Z axis, not servo.

Marmadoc, 2024-04-19:
Thank you. I will try it and let you know.

Marmadoc, 2024-05-05:
.

closed 2024-05-05 by Marmadoc