svenhb/GRBL-Plotter #395: Pen Up/Down via stepper

Marmadoc, 2024-04-30:
I consider to switch from servo to stepper.
I have few questions related to g-code generation
[image]
1.  line 16 G00 Z2 - I understand that it is PU. Z2 is from setup - but what exactly is this number? it is mm? or steps? or ...
it is rapid movement related to GRBL #112 - correct?
2. Line 17 - M3 S100 - why this line is needed  - and what is s number related to.
3. Line 18 - G04 P1 - delay - it use only once in the code. With stepper Pen Up/Dow delay after each pen movement delay is not needed?
4. Line 22 - G01 Z-2.00 F1000 (PD) - if #112 is set to 1000 - what is different between code in line 16 and this one?
5. Line 23 - G01 X47.222 Y32.501 F800 - why here feed is reduced?

6. Last one - svg drag and drop into GRBL is OK. But when I like to reload this file or open different file I have to close program and open it again.  Any different way?

svenhb, 2024-05-01:
1) by default it is mm, you can change the units via G20/G21: https://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g20-g21
2) Lokks like you still have PWM also enabled : https://grbl-plotter.de/index.php?id=form-setup-3#pen-up-down
3) Also from PWM setup, to give the servo time to reach its position
4) Line 16 is pen-up before the first move, line 22 is pen down to start drawing
5) line 22 uses feed-rate for z axis, line 23 uses feedrate for xy axis - check https://grbl-plotter.de/index.php?id=form-setup-3
6) What exactly happens? 
You can reload via menu, via right click on 2D view.
Take care of file loading options, if you want to add new content to current content: https://grbl-plotter.de/index.php?id=form-setup-4#file-loading

[image]
[image]
[image]

Marmadoc, 2024-05-01:
All Make sense.  Right click on 2D view it was really hidden gem.
Thank You

closed 2024-05-01 by Marmadoc

svenhb, 2024-05-02:
Note: after streaming the gcode of an imported graphic, the last used file is the gcode file, not the graphic - any change in import parameters have no effect on "reload"....