svenhb/GRBL-Plotter #396: Stepper motor for pen lift

Marmadoc, 2024-05-05:
I added stepper motor for pen lifter instead of servo motor.
I also add limit switch for pen lifter (Z axis).
In software I can setup Z-Height  - in mm.
I'm looking for gcode generation for example PenUp - move up up to limit - triggered by limit switch and not triggering alarm.
Stepper doesn't have feedback position - so I do not know what is current position (after restarting)
So is here any possibilities to this scenario:
PenUp - move pen to maximum height - limited by limit switch - beginning of code. 
From now on, I can add PenDown PenZero etc.
I'm not talking about homing Z axis. 
Is it possible?

svenhb, 2024-05-06:
>I'm looking for gcode generation for example PenUp - move up up to limit - triggered by limit switch and not triggering alarm.

This is not possible via gcode, except "real" homing. Not sure if it works: if you add "$H" to the header gcode, homing will performed, if enabled: https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle
You could change grbl-firmware to do homing only on Z axis...

Instead you may want to use "tool length offset". I did it here on a micro switch: https://www.youtube.com/watch?v=x5UTHpgsfII
E.g. at 0:33 the tool length will taken via probing: https://grbl-plotter.de/index.php?id=form-probe#toollength

Marmadoc, 2024-05-06:
I did it via $H and config.h modification. It works.
I expected :-) that you have some kind of hidden magic baton to do that by one click.
Tool length - it is new area for me. I will study it soon.
Thx for your help

closed 2024-05-06 by Marmadoc