svenhb/GRBL-Plotter #397: 4 axis tangential knife

abd97khateeb, 2024-05-06:
Hi, When i start the 4D A tangential machine,  A axis rotates correctly in linear shapes, but in curved shapes it rotates differently than the simulation. What is the reason for this?
 video:
https://drive.google.com/file/d/1kFYyj0nKejBGwWmeREaVROWiceNVjwZd/view?usp=drive_link

svenhb, 2024-05-07:
I asked for access to see the video...

abd97khateeb, 2024-05-07:
İ accept

svenhb, 2024-05-07:
Ok, my first thoughts:
- there is no feedrate given in the gcode - that is propbably the problem for the simulator.
Will try to fix that.
- you didn't define a pen up/down-translation (no code before "(PD)"): https://grbl-plotter.de/index.php?id=form-setup-3#pen-up-down
- it is just a simulation, better check on real system
- Why the marked tabs are highlighted?
[image]

svenhb, 2024-05-07:
>you didn't define a pen up/down-translation (no code before "(PD)"

perhaps it is a bug, did you enable "Process meta data for SVG graphic?
[image]

abd97khateeb, 2024-05-07:
No

abd97khateeb, 2024-05-07:
عندما يكون الكود بهذا الشكل G02 x10 y10 I0 J10 A90 فانه ينفذ اولا قسم xy ثم A. اعتقد ان المشكلة في انشاء الكود لأنه اعطا نفس المشكلة في برنامج محاكاة آخر.

abd97khateeb, 2024-05-07:
When the code is like this: G02 x10 y10 I0 J10 A90, it executes the xy section first, then A. I think the problem is in the code generation because it gave the same problem in another simulation program.

abd97khateeb, 2024-05-07:
Is there another way to write code that contains a circular shape?

svenhb, 2024-05-08:
If I use your settings, my code looks like this - with Z axis:
[image]

When you experimenting with settings, you should disable "code add on" - the "old" code stays on top....
[image]

abd97khateeb, 2024-05-08:
Ok thanks i well try.

abd97khateeb, 2024-05-12:
can I convert Z axis to A axis when creating the code because i used Z axis for tangential knife ?

svenhb, 2024-05-12:
If you use Z axis for tangential, you may use a servo (controlled via PWM) to lift the knife. Then you need to select this options:
[image]
[image]

Also note: https://grbl-plotter.de/index.php?id=tangential-knife-1
Units per turn: if using 'Z' as tangential axis, the feedrates will slow down the rotation speed. Using just e.g. 36 units per turn may increase the performance.

abd97khateeb, 2024-05-12:
ok i will try this options, thanks.

abd97khateeb, 2024-05-12:
How can I modify the drawing to combine with the black drawing?
![image](https://github.com/svenhb/GRBL-Plotter/assets/169066818/29e1aff5-add8-4cdd-ba4d-7ef3c672cdeb

svenhb, 2024-05-12:
What do you mean with "combine"? For SVG graphic changes I would use Inkscape...

abd97khateeb, 2024-05-12:
My gcode is difrent than svg shape.
[image]

svenhb, 2024-05-13:
I assume you enabled some additional options, please check code and GUI:
[image]

abd97khateeb, 2024-05-19:
thanke you for helping

closed 2024-05-19 by abd97khateeb