svenhb/GRBL-Plotter #399: Servo is not working

K0niee, 2024-05-16:
Hello everyone,

I have to make a GRBL plotter for school. I made this via HowToMechatronics, after solving some problems I have the last problem that I can't find. When I want to control my servo with the Mcode (M3 S90) so that the color marker is clamped, it does not respond to the commands. It also does not respond when I change a tool. The plotter goes to the tool change point, does everything except the servo. I tested the servo on an Arduino and it works. My servo is connected to the Z+ pin on the CNC motor shield. I would be grateful if anyone could help me.

K0niee

svenhb, 2024-05-16:
If power supply for the servo is ok...
I think you use the wrong grbl firmware. You may try version B or C from here: https://github.com/svenhb/GRBL-Plotter/tree/master/Firmware/hex_files
How to upload the hex file: https://grbl-plotter.de/index.php?id=quick-guide&setlang=en

K0niee, 2024-05-17:
The servo is working, thankyou so much!

closed 2024-05-17 by K0niee