svenhb/GRBL-Plotter #402: How can i set tool offset?

MayHanGroup, 2024-05-22:
Hi,  when i import a svg file with a rect size100mm*100mm,     the gcode example :
G01 X100.000 Y0.000
G01 X100.000 Y100.000
G01 X0.000 Y100.000
G01 X0.000 Y0.000

But when i use a tool diameter 10mm ,  i will get the path  should be like X5.0 ~ X95.0  actually,    i want to find any setting,but not found.    can you help me?

svenhb, 2024-05-22:
For such a simple form, you could use "Gcode from shape" using inside or outside shape option:
[image]

You can also try "Radius compensation" with positive or negative value for the radius of the tool:
[image]

MayHanGroup, 2024-05-22:
oh,it's great!  and also,  is the polygon can use pocket function?   i find the option in "Create Simple shapes" , but  in the form "Create Jog path" i can't find any options,  is jog path not support pocket?    
such like this path
[image]

svenhb, 2024-05-22:
You could try hatch fill for this purpose:
[image]

svenhb, 2024-05-22:
So you would like to have a mix of "Create Jog path" and "Create simple shapes"?
A "Jog path" including pen up/down function to create shapes like in the link you provided.... Have to think about.
Did you try to use Inkscape and copy&paste the shapes?

MayHanGroup, 2024-05-23:
No,  i don't know how to use Inkscape .     when i try to use the "Hatch fill" setting and the "Radius compensation" both , the result looks not as expected .   i there any way to using tool diameter auto when the gcode generate?   i means like the settings in the form "Create Simple shapes" .   
[image]
[image]
[image]

svenhb, 2024-05-23:
>i there any way to using tool diameter auto when the gcode generate? 

Not yet. The problem would be, how to determine if the offset should be inside or out side the shape. You must follow a "design rule" when making your drawing (CW and CWW path). What tool do you use (if not inkscape or LibreCAD - both I like to use).
E.g. a picture frame: for the outside contour, the offset must be outside, for the inside contour inside. If the tool-offset is to the left, the outer contour must be CW and the inner CCW.
[image]

MayHanGroup, 2024-05-23:
ok i know.  thank you for your help.  @svenhb

closed 2024-05-23 by MayHanGroup