svenhb/GRBL-Plotter #404: Height map not working

elkor77, 2024-05-27:
Hello, my problem is the following: I try to load a height map in a file that has 5150 lines but the attempt takes a while and gives an error. It is a small vector, it only measures 75x53 millimeters. I am using an Arduino uno R3 with a PC with a Core I7 processor and 12 Gb of Ram.

This is the vector
[image]

This is the height map
[image]

This is what comes out after a long time
[image]

And this too
[image]

I share the Grbl file and the height map file with you
[Grbl and Map.zip](https://github.com/svenhb/GRBL-Plotter/files/15462371/Grbl.and.Map.zip)

svenhb, 2024-05-28:
Strange: I took your data and it works...
But after setting all to default (like on new installation) it failed once....
I assume the default of this setting is wrong, can you change this value (e.g. to 0.4) and the "Apply changes", then try again...
[image]

elkor77, 2024-05-28:
> Extraño: tomé tus datos y funciona... Pero después de configurar todo a los valores predeterminados (como en una nueva instalación) falló una vez... Supongo que el valor predeterminado de esta configuración es incorrecto, ¿puedes cambiar este valor (por ejemplo, a 0,4)? ) y "Aplicar cambios", luego inténtalo de nuevo... ![imagen](https://private-user-images.githubusercontent.com/19973251/334317279-0e17e87b-0d66-4209-ae68-a3cd255568f9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY5Mzk2NzgsIm5iZiI6MTcxNjkzOTM3OCwicGF0aCI6Ii8xOTk3MzI1MS8zMzQzMTcyNzktMGUxN2U4N2ItMGQ2Ni00MjA5LWFlNjgtYTNjZDI1NTU2OGY5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI4VDIzMzYxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM0ZWE0NDIyMzJiMTYzMDVhYjFlYzcyNGRhOWZlNmQ0NWEyNDUzNTA1ZGUyNzMyZGEzZDc5M2RjNTA4NDM2NWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.iU-E-llm3r1wTaR3om73HQyFYODDjIPdzc5nu_Z3oII)

Thank you very much, I changed the value to 0.50 as it is in the image you sent and it worked for me. Excellent software, I'm starting to use it and I really liked it. Congratulations on your work. Thank you

closed 2024-05-29 by elkor77