svenhb/GRBL-Plotter #405: Circles not round after Plotting

GenerativeOlli, 2024-06-02:
**Describe the bug**
After PenPlotting an image importet as an svg-file, circles are composed of smooth arcs and straight lines. This also puts the spacing off.

**To Reproduce**
Steps to reproduce the behavior:
1. I drag and dropped the svg file into the plotter window
2. I accept the conversion option
3. Everything looks fine and I start the plotter by sending the gcode
4. Circles are off.

**Expected behavior**
A clear and concise description of what you expected to happen: Seeing crisp images.

**Screenshots**
The Picture shows on the left: the result with a pen that needs pressure to operate so it created an offset and my pen holder was weak and gave it some room to flex. On the right with black ink: I redesigned the entire penholder, so it is very ridget now and switched to a fineliner without pressure. Minimal offset it there due to the paper only fixed on one side.

[image]
[Streights and Circles.txt](https://github.com/user-attachments/files/15524532/Streights.and.Circles.txt)
[image]

Thank you for looking into this =) Keep up the great work!

GenerativeOlli, 2024-06-02:
Note: The txt file contains a gcode I generated later the same way. I dismissed the original when I closed the window.

GenerativeOlli, 2024-06-02:
please ignore. I found the problem being the way my design script calculated stuff.

svenhb, 2024-06-02:
Ok, so GRBL-Plotter works as expected?

closed 2024-06-02 by svenhb

GenerativeOlli, 2024-06-02:
Yes it does! Thank you for the great work!