svenhb/GRBL-Plotter #406: The current shortest path sorting algorithm is not optimal

lianzaozi, 2024-06-03:
The current shortest path sorting algorithm is not optimal

Plotter software sorting results
[image]

Sorting results from other software
[image]

Test files

[666.zip](https://github.com/user-attachments/files/15527373/666.zip)

svenhb, 2024-06-03:
It depends on the start-point. In the current version sorting starts with most upper-left object (always).
In the next version you can decide where to start:
[image]

svenhb, 2024-06-03:
Result with new option - start at lower-left edge:
[image]
[image]

lianzaozi, 2024-06-03:
Thank you for your reply. I appreciate your hard work and look forward to the sorting effect in the next version.
Can you add a line of information below to list the total length of the empty movement path and the total length of the machining path.
[image]

closed 2024-12-15 by svenhb