svenhb/GRBL-Plotter #411: Crash when running under Proton in Linux

ArcaEge, 2024-06-27:
**Describe the bug**
I followed the steps in the Linux install guide, however when I attempt to connect to GRBL, the whole program crashes. This only occurs on a valid serial port with a serial device connected, however it also crashes when there is no GRBL running (it crashed with an Arduino running blink in my testing).

**Details**
System: Fedora Workstation 40 running GNOME/X11
Proton: Crashes under both 9.0-2 and experimental (haven't tested other versions)

**To Reproduce**
Here is a screen recording to make things easier:
https://github.com/svenhb/GRBL-Plotter/assets/40526225/20231eb7-95fc-4399-b7dd-5cf144b367d2

**Expected behavior**
It should connect to GRBL and not crash.

Here is a paste of the log (I had to truncate some parts in the middle)
https://pastebin.com/hrnPjzCu

svenhb, 2024-06-28:
Sorry no idea about Linux.
Did you select a valid port first? Is COM10 your port?
[image]

ArcaEge, 2024-06-28:
OK, would you like me to do additional testing? I can connect it to a GRBL machine that I know did not cause a crash in Windows

svenhb, 2024-06-29:
You may have look into  (or upload) the logfile of GRBL-Plotter. 
Menu - About - open the "Data Path", go to subfolder logfiles and check "logfile.txt".

ArcaEge, 2024-06-29:
OK, I've had to reinstall my system because I corrupted my OS partition trying to resize it, I'll have a look at this soon.

closed 2024-12-15 by svenhb

ArcaEge, 2024-12-18:
Sorry for forgetting about this... I've resorted to running Windows in a VM with USB passthrough using `gnome-boxes` instead since that was easier to set up. If I manage to find time, I'll modify `GRBL-Plotter-Linux.md`, listing how I set it up and submit a PR in the next few days.

svenhb, 2024-12-18:
Ok, thanks