svenhb/GRBL-Plotter #413: Manually set Tool Coordinates

stujoult, 2024-08-10:
**Describe the solution you'd like**
Form to manually set current Tool Coordinates

**Describe alternatives you've considered**
Something like UGS
[image]

svenhb, 2024-08-10:
Good idea, need to think about where and how to implement...
- click on digital read out, to show edit box like in your example? Perhaps not intuitive
- right click on joystick control to open extra form with controls?
- or dropdown box below joystick to show controls?
[image]
[image]

stujoult, 2024-08-11:
Something like this maybe?
[image]

svenhb, 2024-08-11:
I put it into the control/ jogging area.
Need some tests for other new features, before uploading.
[image]
[image]

stujoult, 2024-08-11:
I think it might confuses people. Instead of jogging to specific location like what user would expect, it only moves tool pointer.
Also that STOP Jogging is a good feature. My monkey fingers keep misclicking 5 instead of 1.

svenhb, 2024-08-11:
>I think it might confuses people. Instead of jogging to specific location like what user would expect, it only moves tool pointer.

I don't get it. In my eyes: Tool-pointer = tool position = work coordinates of the machine...
Move to zero [image] is the same as this button [image]
To move to X 100 you put 100 here and press "Move to".
[image]
 

 

>My monkey fingers keep misclicking 5 instead of 1.

You talk about the joystick control? Did you know you can resize it`?
[image]

stujoult, 2024-08-12:
> I don't get it. In my eyes: Tool-pointer = tool position = work coordinates of the machine...

I was talking about feature to set virtual tool coordinate without moving the actual tool coordinate. Like when my spindle currently at X10Y75, [Zero XY] only set tool pointer to X0Y0 and spindle position remains unchanged.

> You talk about the joystick control?

Yes. But it is modern laptop touchpad problem. Instead of just click, it will slightly moves cursor and click. My brain is already hardwired to good old touchpad design.
[image]
The good old design
[image]

svenhb, 2024-08-12:
Got it :-)
Regarding the touchpad: I can't work with it and that's why I always have a mouse connected...

stujoult, 2024-08-19:
No [Set new origin] for rotary axis?
And missing [STOP Jogging] button.
Thanks in advance.
[image]

svenhb, 2024-08-19:
Did you connect a controller? 
"Stop Jogging" only appears if connected and grbl version >= 1.1
Set new origin A axis appears if connected  and axis available

stujoult, 2024-08-20:
Well I only tested it on 0.9. Will test it on Mega later.

closed 2024-08-20 by stujoult