svenhb/GRBL-Plotter #414: Wrong Dimension when open this file

zengfanfan, 2024-08-29:
**Describe the bug**
When open a g-code file, the minimal Z value is not right. ( should be 0, but we got 10)

**Screenshots**
[image]

**the g-code file**
```G-Code
( *** ZengXYZ Plotter v82M *** )
( Generated by gcode_plotter at 2024-08-30 03:02:56.633885 )
( Oxy = <88,88> , Z = 0 ~ 10 (safe=80) )
( MoveSpeed = 12000, FillSpeed = 6000 )
( Pen.default = 3 )
( Pen.1: width=1.0, offset=Pt<-52,0> )
( Pen.2: width=1.0, offset=Pt<-26,0> )
( Pen.3: width=2, offset=Pt<0,0> )
( Pen.4: width=1.0, offset=Pt<26,0> )
( Pen.5: width=1.0, offset=Pt<52,0> )
( File: 1.gcode )

( <Header> )
; pull up Z
G90 G1 Z80 F6000
( </Header> )

G90 G21 ; absolute distance mode, units: mm
G1Z80F12000 ; safe Z
G1X88Y88F12000 ; goto Oxy

( <Group id="1"> )
( </Group> )

( <Group id="2"> )
( <Pen id="3" width="2" offset="0,0" /> )
M6T3
G1X136.9Y101.3Z0
M3S1
G1X139F6000
M3S0
M5
G1Z10F12000
G1X163.1Y111.2
G1Z0
M3S1
G1X163.1Y111.2F6000
G1X164.7Y112.8
M3S0
M5
G1Z10F12000
G1X174.6Y136.9
G1Z0
M3S1
G1X174.6Y136.9F6000
G1X174.6Y139
M3S0
M5
G1Z10F12000
G1X164.7Y163.1
G1Z0
M3S1
G1X164.7Y163.1F6000
G1X163.1Y164.7
M3S0
M5
G1Z10F12000
G1X138.9Y174.6
G1Z0
M3S1
G1X138.9Y174.6F6000
G1X137Y174.6
[... 2085 lines cut ...]
; importFlipY: false
; exportFlipX: false
; exportFlipY: false
; resolution: 10
; fillSpeed: 6000
; moveSpeed: 12000
; pens: 
;   - |-
;     {"default":false,"color":"000000","offset":"-52,0","width":1.0}<end!
;   - |-
;     {"default":false,"color":"FFFFFF","offset":"-26,0","width":1.0}<end!
;   - |-
;     {"default":true,"color":"CF3B31","offset":"0,0","width":2}<end!
;   - |-
;     {"default":false,"color":"4CAF50","offset":"26,0","width":1.0}<end!
;   - |-
;     {"default":false,"color":"2196F3","offset":"52,0","width":1.0}<end!
; header: |-
;   ; pull up Z
;   G90 G1 Z80 F6000
;   <end!
; footer: ""
( </Settings> )

```

svenhb, 2024-08-29:
Ok will check.
You can also upload files, instead of copying the content....

stujoult, 2024-09-02:
Reproduced in older version.
It treats `z0` as non number.
[image]

svenhb, 2024-09-02:
I don't remember why I did it this way: dimension was only updated for Z, when z-position is not actual grbl z-position:
`if (posWasSet && (newLine.actualPos.Z != Grbl.posWork.Z)) { xyzSize.SetDimensionZ(newLine.actualPos.Z); } `
Is fixed now, I make a new release after finishing other changes...

zengfanfan, 2024-09-07:
ok. looking forawd to it.

svenhb, 2024-12-15:
Check new release: https://github.com/svenhb/GRBL-Plotter/releases

closed 2024-12-15 by svenhb