svenhb/GRBL-Plotter #415: Replace G2/3 by lines function

yang-shaoxiong, 2024-09-04:
The Replace G2/3 by lines in the G-Code Transform in 1.7.3.1 will report an out of memory error when used, but the actual computer still has a lot of memory. What is the reason for this?

svenhb, 2024-09-04:
What are your settings here? 
To reduce data amount:
- Try to decrease "Number of line segments in curves"
- Try to increase "Step width on circumference of arcs"
[image]

yang-shaoxiong, 2024-09-04:
Isn't this a file import? I'm
[image]
 talking about the Replace G2/3 by lines function in the G-Code Transform menu

svenhb, 2024-09-04:
I checked the code, it will use this parameter - also for [Menu - G-Code Transform - Replace G2/3]: 
[image]

yang-shaoxiong, 2024-09-04:
Ok, thanks for your reply

svenhb, 2024-09-04:
You may experiment with a simple G-Code and check the amount of lines after replacing G2/3 with different step-widths...
With step width 0.01 more than 6000 code lines, 
with step width 0.5 less than 150 code lines and it is acurate enough.
```
(<Figure Id="1" Geometry="Shape" >)
G00 X-10.000 Y0.000  
G01 Z-1.000 F400 
G02 X-10.000 Y0.000 I10.000 J0.000  F500 
G00 Z2.000 
(</Figure>)
```

yang-shaoxiong, 2024-09-05:
That's all. Thank you for your explanation.

yang-shaoxiong, 2024-09-05:
Hello, can we pay you to customize some features for us? It can be modified on the basis of the original.

svenhb, 2024-09-05:
No paid orders. List the features, let's see what I can implement and donate ;-)

yang-shaoxiong, 2024-09-06:
I want to customize the path and the cutting point, and use PLT to represent, not G-Code.

svenhb, 2024-09-06:
Then it is probably easier to use Inkscape https://www.inkscape.org/ (very good for path editing) and export there to *.plt

yang-shaoxiong, 2024-09-06:
Good. Thanking

yang-shaoxiong, 2024-09-21:
Hello author, I try to use Win32_ API for USB communication in GRBL project, but can not get USB device, can get it in other projects. Is this what permissions are set for the project?

svenhb, 2024-09-21:
No idea what the problem could be. Longer time ago I run GRBL-Plotter on Win-XP but PC was too weak...
Did you install the correct USB driver for your hardware? (e.g. CH 340 -> https://www.google.de/search?q=arduino+ch340)

closed 2024-12-15 by svenhb