svenhb/GRBL-Plotter #416: G2/G3 don't recognize P(full circle) parameter

zengfanfan, 2024-09-13:
#### How ?
open the attached file, which includes a 540°(1.5 circle) arc.

#### Bugs
1: the preview image only shows 0.5 circle;
2: it runs 0.5 circle in simulation, but 1.5 circle in real machine.

**preview:**
[image]

**test file:**
[bug.gcode.txt](https://github.com/user-attachments/files/16988009/bug.gcode.txt)

svenhb, 2024-09-13:
Too bad, the P-value in `G2 X165.0 Y132.0 I44 J22 P2` is not supported by original grbl (https://github.com/gnea/grbl) and also not supported by GRBL-Plotter.
Original grbl gives this error: 
```
< error:36 	Unused value words found in block.
! Last processed 'G2X165.0Y132.0I44J22P2'  line-Nr:24

```
Can't you use a pre-processor to avoid the P-word?

zengfanfan, 2024-09-13:
ok.

closed 2024-09-13 by zengfanfan

svenhb, 2024-12-15:
Perhaps I could fix it for the 2D-View: https://github.com/svenhb/GRBL-Plotter/releases