svenhb/GRBL-Plotter #420: Triggering automations from external source e.g. python script

Dan-Zed, 2024-10-02:
**Describe the solution you'd like**
I think using a simple button to the probe pin is brilliant, but for my workflow I would like to trigger an automation from my computer but external to GRBL-Plotter.

Alternatively, allow GRBL-Plotter to wait for the clipboard to be populated i.e., not fail if it's empty.

**Describe alternatives you've considered**
I'm a novice at coding. My first thought was that GRBL-Plotter could wait for the clipboard (very cool feature, btw) to be populated with a text but it seems to fail if the clipboard is empty.

So far I've explored ways to interface with GRBL-Plotter or grbl. I've been able to use a python script to trigger the grbl probe pin, but of course that requires sharing port access which doesn't work. I thought maybe one of the custom buttons could be used to trigger an automation (then activated by hotkey) but that also doesn't work. I investigated the extensions and scripts and I don't think either of those help me. 

**Additional context**
The short story of my (art) project is: I will be generating graphics or texts programmatically in response to an input. This is largely done in python. I want the final result of this program to trigger my plotter to generate the graphic/text, but completely automated without intervention, except for the initial upstream input by the user to my python program. I'd be happy to share the whole project if/when I complete this :) 

If there is no internal solution to this, the only other thing I can think of is a hardware hack where I connect another controller/arduino/switch to emulate the button. But that's not ideal.

svenhb, 2024-10-03:
Sounds good, short answer for the moment: GRBL-Plotter already checks the clipboard for changes, implemented for the extensions:
[image]

https://grbl-plotter.de/index.php?id=form-extenions&setlang=en

svenhb, 2024-10-03:
Check the folder data\extensions for examples: 
You can get GRBL-Plotter values via the registy:
reg_key="HKEY_CURRENT_USER\\SOFTWARE\\GRBL-Plotter\\GCodeSettings\\";
reg_key="HKEY_CURRENT_USER\\SOFTWARE\\GRBL-Plotter\\Position\\";

To trgigger GRBL-Plotter to read the clipboard - set this registry flag:
function set_GRBL_Plotter_Update()
{	var reg_key="HKEY_CURRENT_USER\\SOFTWARE\\GRBL-Plotter\\";
	try 
	{	GRBL_Plt_decimals    = WshShell.RegWrite(reg_key + "Update",1,"REG_DWORD");	}
    catch (err) {}
}

svenhb, 2024-10-03:
So finally I think, to realise your idea, just one point is missing: triggering the process automation by change of the shown registry-key. 
 - wait for the registry key to change,
 - if triggered, load file from clipoard and import
 
Will try to implement this

I already made a similar automation with a website, but without polling a status, 
just trigger the website  (https://grbl-plotter.de/fishdraw/) to create content, then read the clipboard, but, it stucks with the first loop:
 
```
<?xml version="1.0" standalone="yes"?>
<NewDataSet>
  <ProcessItem>
    <Command>Load URL</Command>
    <Value>https://grbl-plotter.de/fishdraw/</Value>
    <Info>Open web site [Value=URL]</Info>
  </ProcessItem>
  <ProcessItem>
    <Command>Paste clipboard</Command>
    <Value />
    <Info>Paste content from clipboard</Info>
  </ProcessItem>
  <ProcessItem>
    <Command>G-Code Stream</Command>
    <Value />
    <Info>Send G-Code from editor to machine</Info>
  </ProcessItem>
  <ProcessItem>
    <Command>G-Code Send</Command>
    <Value>(^2G0G91X130;M8G4P1;M9)</Value>
    <Info>Send G-Code to machine, [Value=g-code commands or macro-file] seperate single command lines with ';'</Info>
  </ProcessItem>
  <ProcessItem>
    <Command>Jump to</Command>
    <Value>1</Value>
    <Info>Jump to line number, [Value=&lt;linenumber&gt;;&lt;repetitions&gt;]</Info>
  </ProcessItem>
</NewDataSet>
```

Dan-Zed, 2024-10-04:
Thanks for acknowledging this so quickly.

Just to double check the expected behavior:

With an empty clipboard, the fishdraw automation you shared opens up the website then immediately proceeds to the paste clipboard step where it stops. A rerun with the copied svg is then loaded on the second iteration. 

The reg_key approach actually gets me very close to what I need if also change the Start reg key to trigger streaming. Together with the right custom settings .ini (disabling the "show dialog on import" prompt) I think this will complete the workflow. 

I suppose that waiting for the reg key change in the automation would allow some additional control and features, but the above approach I think should suffice for my use case. 

As soon as I have a chance, I will test out the full workflow and report back.

svenhb, 2024-12-15:
Triggering via registry key is implemented: https://github.com/svenhb/GRBL-Plotter/releases

closed 2024-12-15 by svenhb