svenhb/GRBL-Plotter #422: Version 1.7.4.0 of the plotter software reported an error in the source code when using text to generate the GCode function

lianzaozi, 2024-11-05:
[image]
[image]
[image]

svenhb, 2024-11-06:
Can you change the marked line
[image]
 to
`                if ((textFont != null) && (textFont.Size != null) && (textFont.FontFamily != null))`
and check if exception disappears?

lianzaozi, 2024-11-06:
There are other errors
[image]
[image]

svenhb, 2024-11-06:
The other error appeared after doing the change?
              `  if ((textFont != null) && (textFont.Size != null) && (textFont.FontFamily != null))`

lianzaozi, 2024-11-06:
YES

lianzaozi, 2024-11-08:
Hello, how do I open the second window connected to the com port, which is the window shown in the picture
[image]

svenhb, 2024-11-08:
[image]

If you want the window on program start:
[image]

lianzaozi, 2024-11-08:
Thank you very much. Following your instructions, I have already opened this window

lianzaozi, 2024-11-08:
When using Gcode created from text, the source code will run incorrectly, but running the GRBL-Plotter.exe file under the debug folder will work fine, I was surprised.

[image]

svenhb, 2024-11-09:
It works because of the "try" / "catch" block, which then handles the error you got.

lianzaozi, 2024-11-09:
Is there any good way to solve the problem of source code debugging errors? I tried but couldn't find a good solution

svenhb, 2024-11-09:
I don't know, I don' t get this error... Perhaps you need enable the other checkbox or disable the first....
[image]

lianzaozi, 2024-11-09:
Thank you, I have one more question
Where can I set the distance and speed of movement after clicking the JOG button each time?
Every time I click the jog button, the motor moves 10mm in a fixed position
[image]

svenhb, 2024-11-09:
If a controller is connected and the buttons are enabled - just right click to get a selection list:
[image]

The choices you can define here:
[image]

lianzaozi, 2024-11-10:
[image]

After unchecking "Optimize code", using the function of generating Gcode from text will result in an error If checked, there will be no errors

svenhb, 2024-11-10:
The main thing is that the error does not occur if the program is started regularly and not via VisualStudio

closed 2024-12-15 by svenhb