svenhb/GRBL-Plotter #423: Set delayed laser on and early laser off

lianzaozi, 2024-11-10:
[image]

When laser cutting laser energy should not be in the motor acceleration and deceleration stage output set energy, should this time the motor is slow, laser energy concentrated in a point will lead to overburning phenomenon, solutions :1. When the laser is turned on, the laser energy slowly increases to the set value, when the laser is turned off, the laser slowly decreases to 0(as shown in the figure, when the laser cutting machine cuts from point A to point D, the motor is accelerated from A to B, at this time the laser energy should slowly increase to the set value, when the motor moves to point B exactly the laser energy increases to the set value, this situation is the most All right; When the laser runs to point C, the laser energy should be slowly reduced, and when the laser position moves to D, the energy is reduced to 0, which is the best effect.

2. When the laser is turned on, the laser is later than the motor is turned on, when the laser is turned off, the laser is turned off earlier than the motor is turned off in advance, and the laser time and the delay in turning on the laser time can be set (as shown in the figure). When the laser cutting machine is cut from point A to point D, the motor is accelerated from point A to B, and the laser energy should be opened at the motor movement to point B, which is the case The best; When the laser runs to point C, the laser energy should be reduced to 0.)

lianzaozi, 2024-11-10:
[image]
When cutting to the four corners of the rounded rectangle, which are 1/2/3/4, the energy of the laser should be appropriately reduced. This is because the total speed of the motor's x-axis and y-axis arc interpolation will decrease at the rounded corners. If the laser energy is not reduced at this time, it will lead to over burning of the four corners, especially when cutting very thin materials. The situation of over burning is more obvious if the energy is not reduced

svenhb, 2024-11-10:
I think the M4 laser mode should cover this: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode:
M4 Dynamic Laser Power Mode:
Dynamic laser power mode will automatically adjust laser power based on the current speed relative to the programmed rate.

lianzaozi, 2024-11-13:
thank youThank you for your reply. These pieces of information are very useful to me

lianzaozi, 2024-11-16:
What is the function of the "Displays the pen down path nodes" option?
[image]

svenhb, 2024-11-16:
Then a little cross marks each coordinate given in the G-Code:
[image]

lianzaozi, 2024-11-16:
Thank you for your guidance. Thank you

closed 2024-12-15 by svenhb