svenhb/GRBL-Plotter #426: Interface text goes small when import SVG o DXF

Toaro, 2024-11-17:
When I open the software all appear to be OK, with readable fonts in normal size, if I load a JPG file for conversion still all is normal, but if open a SVG or DXF file, all the interface text becomes very small

**To Reproduce**
Steps to reproduce the behavior:
1. Open GRBL Plotter soft
2. Drag and drop SVG file, just when you drop, the text interface change

**Expected behavior**
The text interface must to stay in the same size

**Screenshots**
Here some screenshots. I include some desktop icons and text for comparison purposes. I don't move the program window when i load the SVG file, but the window also change the position, as you can see with the desktop icons reference

That is open without touch anything.
[image]
That is when I drop any SVG to the window.
[image]
As you can see the gcode window text becomes very "unconfortable" to read because lost all the space between the letters and lines. I solve the lines space changing in the configuration menu the line distance, but that becomes in a very large distance when the interface is ok.
[image]

Thanks for your work.

svenhb, 2024-11-17:
I can't fix this HDPI problem, but this may help: https://grbl-plotter.de/index.php?id=quick-guide#highdpi

Toaro, 2024-11-17:
Thanks!!!. That fix the problem.

Thanks again for the software, your work, and your time. Great

closed 2024-12-15 by svenhb