svenhb/GRBL-Plotter #427: Please add Text weld feature

AR4672, 2024-11-17:
Add text weld option in G Code creation (create text).

While creating gcode i found that alphabet gets detached in text mode, this creates a problem in vinyl cutting. 

[image]

svenhb, 2024-11-17:
This not easy to implement.
I would use Inkscape to create the text,
- set "stroke" to black and "Fill" to none
- then "object to path"
- ungroup
- Edit node path to your whixhws
- copy whole drawing and paste to GRB-Plotter (with drag knife option enbaled)

AR4672, 2024-11-20:
Thanks for the reply :)

Well, I have used this method and works nice, but I thought if create text would have the option to weld then it would have been an easy task to create precise gcode without any hassle.        
 
BTW that drag knife option is like a life saver for me.  

Actually, I have transformed a printer into cnc plotter but continuously facing precision issue, I thought it was a design flaw but later got to know about that offset setting, that's only present in your GRBL-Plotter.

All thanks to you ....

closed 2024-12-09 by AR4672