svenhb/GRBL-Plotter #428: The shortest path sorting of text trajectory is not optimal

lianzaozi, 2024-11-27:
Sorting of plotters
[image]

Sorting of other software
[image]

svenhb, 2024-11-27:
Perhaps you used wrong sorting option?
I would use this settings:
[image]

lianzaozi, 2024-11-27:
Thank you very much. According to your instructions, the text sorting problem has been solved now. Can you explain the differences between these options for sorting?
[image]

svenhb, 2024-11-27:
This is just the starting point for sorting, relativ to the G-Code or actual tool position ("actual pos").

I think in your example you checked "Sort the objects by size, largest comes last"

lianzaozi, 2024-11-28:
Due to my mistake, I used the default sorting rule and indeed selected 'Sort the objects by size, large returns last'
Thank you very much for your guidance

lianzaozi, 2024-11-28:
The plotter software does not display the trajectory after reading the image and generating GCode

[image]
[test file.zip](https://github.com/user-attachments/files/17943127/test.file.zip)

svenhb, 2024-11-28:
I thing you useed strange option for "pen up/down" translation:
Your Z never go to minus
[image]

Also you hide "above 0" paths:
[image]

closed 2024-12-15 by svenhb