svenhb/GRBL-Plotter #429: Error `Tool table not found` during `Import Machine Settings`

isukces, 2024-12-19:
# Issue: Error `Tool table not found` during `Import Machine Settings`

## Problem Description
The `Import Machine Settings` feature causes the error `Tool table not found`. This issue occurs because the path to the tool file is incorrectly processed. The path is concatenated with `Datapath.Tools` using a backslash (`\`). However, the `.ini` file already contains the full path, so concatenating results in an invalid file name.

### Steps to Reproduce
1. Attempt to import machine settings using the `Import Machine Settings` feature.
2. Observe the error: `Tool table not found`.

### Cause
The path to the tool file is constructed incorrectly as:
```csharp
Datapath.Tools + "\\" + tmpstr
```

### Solution
Modify the code in the IniFile.cs file to use `Path.Combine` instead of manual string concatenation.

### Correct Code:
```csharp
Path.Combine(Datapath.Tools, tmpstr)
```

### Additional Notes
Using Path.Combine ensures proper handling of paths regardless of the platform and avoids issues caused by incorrect concatenation.

svenhb, 2024-12-19:
Thanks, I will test

buddasticks, 2025-01-17:
i also have the exact same issue trying yo load use case

svenhb, 2025-01-17:
I made changes, but forgot to make a new release. Can you check this version?
https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1751.exe

buddasticks, 2025-01-18:
Will test when home .. my work around was just to remove the called for tools.csv from the saved caseuse.ini

buddasticks, 2025-01-23:
It appears to work

svenhb, 2025-02-26:
Now available in offical release: https://github.com/svenhb/GRBL-Plotter/releases/tag/v.1.7.5.2

closed 2025-02-26 by svenhb