svenhb/GRBL-Plotter #430: svg tool path seems to disappear upon applying the rotate 90 degrees gcode transform

anandijain, 2025-01-08:
Firstly, thank you for the incredible program. It is incredibly useful as a first-time CNC builder to have something to easily generate gcode.

For some reason my path disappears when I apply a rotation transformation. If I simulate the path I still see the green outline. 

Help greatly appreciated, thank you

Here is a video of what I am seeing:
https://github.com/user-attachments/assets/4f96493e-b7c9-4a3f-b32e-4b18f19fc2f0

svenhb, 2025-01-09:
I could not reproduce the error.
Perhaps you can upload your machine settings? (Menu - File - Export machine Settings).
Did you just Import one image and some SVGs and tried rotation, or did you try also other transforms before rotation?

If you want to transform an imported image, better switch off "compress code":
[image]

If you don't want to add imports to the current workspace, you may switch off this option:
[image]

svenhb, 2025-04-12:
https://github.com/svenhb/GRBL-Plotter/releases

closed 2025-04-12 by svenhb