svenhb/GRBL-Plotter #431: A few questions on process automation and controlling grbl-plotter with a webserver

anandijain, 2025-01-14:
I have been using GRBL-Plotter a lot recently and it is incredible. 

I am trying to make my plotter get to the point where I could trigger a print remotely by hosting a webserver on my computer and then having GRBL run a custom script/process when it receives a request.
 
I found process automation close to what I think would make this work, but I have a few questions. The items with an asterisk have a question below.  

My standard workflow so far is:
1) take an image and convert to svg in inkscape 
2) render the svg to png in inkscape 
3) *load png into GRBL-Plotter with specific settings, vectorization, BW threshhold value etc 
4) *mirror X 
5) *scale XYY to 170 
6) offset to 7;2;2
7) generate gcode
8) home
9) stream 

note: in a few cases i found that rendering jpg -> svg -> png in inkscape can cut down code size (a few times this has caused issues for me in GRBL-plotter.) Not relevant to this issue.

Questions:
1) When doing a "Load" in "process automation" for a "Convert Image" operation, do I just need to save my own `ini` file? If so, is there a documented list of all the things I need to set? Is everything I need to specify exported when I do "Export machine Settings"? I see a setting `Image detection threshold=128` in my settings. 
2) Is there a way to automate the gcode transforms, I in particular need to do a "Mirror X" operation to get everything to work
3) My printer has dimensions of max 240 in X and 170 in Y. If an image has an aspect ratio where the X dimension is much bigger than Y, then I need to scale by XYX 240. If otherwise, then I want to scale to XYY 170. Essentially: "make the biggest image possible without going past the max bounds".
4) Do you see a way to trigger GRBL through a webserver? I was thinking I could always hack it by doing some sort of automated mouse clicking. Would appreciate your thoughts? 
5) When I run my custom process, this tool change thing pops up and requires manual intervention, how can I disable this? 
[image]

This is what my custom process looks like currently:
[image]

This is the result of "Export Machine Settings" 
[GRBL-Plotter_20250113_155837.txt](https://github.com/user-attachments/files/18403606/GRBL-Plotter_20250113_155837.txt)

Thank you very much, this software has been an absolute joy to use!

anandijain, 2025-01-14:
[image]
One last question is how I can avoid this dialog? Even if I add a separate "Load" command with my INI, the gcode doesn't seem to generate unless I actually click the button to "Create G-Code" manually.

svenhb, 2025-01-14:
Your workflow sounds interesting.
Some of your problems could be avoided if you adapt your workflow - I think:
1. take an image an convert to svg in inkscape
2. mirror X in inkscape
3. scale XY in inkscape
4. load svg into GRBL-Plotter with specific settings,
5. offset to 7;2;2
6. home
7. stream

To your questions:
The "old" convert image dialog can only be avoided, if you select "Automatic vectorization" for your desired image-format: 
[image]

1. up to now "Convert Image" does'nt store any settings in an "INI" file
2. up to now Process automation can't mirror X or Y (but it is easy to implement)
3. good idea to set limits on where the content has to fit and calculate the scaling accordingly.
4. I have no clou about web-server technic and don't know how GRBL-Plotter could be triggered...
Up to now you could trigger by a registry-key change or by pasting content into the clipboard.
5. To avoid tool change message, just disable tool change option in setup or "Convert image":
[image]

svenhb, 2025-04-12:
I hope it works now as expected....

closed 2025-04-12 by svenhb