svenhb/GRBL-Plotter #432: Estimation time

Atlan4, 2025-01-31:
**Describe the bug**
The window where the G-code dimensions are displayed is small, it does not display the important data for me. Time. I have to click on the window and move the cursor down to display the time. Thank you

Increase the size of the window downwards to display the time.

svenhb, 2025-01-31:
You can resive the left part of the window:

[image]

svenhb, 2025-02-01:
Check new test version: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1751a.exe

Atlan4, 2025-02-02:
It looks good. Thanks. I just don't like the installation, and I couldn't find any folders where I could move my settings. As with other .zip versions

svenhb, 2025-02-26:
Now available in offical release: https://github.com/svenhb/GRBL-Plotter/releases/tag/v.1.7.5.2

closed 2025-02-26 by svenhb