svenhb/GRBL-Plotter #433: Tangential Knife doesn't work when using A-axis as Name of Tangential Knife

asoka777, 2025-02-03:
I attempted to use the A-axis for the Tangential Knife since I need the Z-axis for lifting the knife to cut a 5 mm foam board. However, it doesn't seem to be working. I created a rounded rectangle using the G-Code Creation menu, but when I run it, I encounter error messages.

[image]

----------------------------------------------------------------
[Start streaming - no echo at 2025-02-02 22:53:09 ]
[Streaming Protocol: Character-Counting | RX-mode: async.]

[Save last pos.: X=   19.998 Y=    0.000 Z=   -1.800]
Progress: 30%  line:    14   time:22:53:09

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G01A90.000'  line-Nr:18
Try to continue, err-count:1

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G00X-1.500Y-61.500A90.000'  line-Nr:20
Try to continue, err-count:2

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G01X-1.500Y-18.500A90.000F200'  line-Nr:22
Try to continue, err-count:3

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G02X18.500Y1.500I20.000J0.000A0.000'  line-Nr:23
Try to continue, err-count:4
Progress: 40%  line:    19   time:22:53:10

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G01X61.500Y1.500A0.000'  line-Nr:24
Progress: 50%  line:    24   time:22:53:11
Try to continue, err-count:5

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G02X81.500Y-18.500I0.000J-20.000A-90.000'  line-Nr:25
Try to continue, err-count:6

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G01X81.500Y-61.500A-90.000'  line-Nr:26
Progress: 55%  line:    26   time:22:53:11
Try to continue, err-count:7

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G02X61.500Y-81.500I-20.000J0.000A-180.000'  line-Nr:27
Try to continue, err-count:8

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G01X18.500Y-81.500A-180.000'  line-Nr:28
Try to continue, err-count:9

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G02X-1.500Y-61.500I0.000J20.000A-270.000'  line-Nr:29
Try to continue, err-count:10
Progress: 40%  line:    19   time:22:53:11

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G01A90.000'  line-Nr:32
Try to continue, err-count:11

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G00X-1.500Y-61.500A90.000'  line-Nr:34
Try to continue, err-count:12

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G01X-1.500Y-18.500A90.000F200'  line-Nr:36
Progress: 75%  line:    36   time:22:53:11
Try to continue, err-count:13

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G02X18.500Y1.500I20.000J0.000A0.000'  line-Nr:37
Try to continue, err-count:14

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G01X61.500Y1.500A0.000'  line-Nr:38
Try to continue, err-count:15

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G02X81.500Y-18.500I0.000J-20.000A-90.000'  line-Nr:39
Try to continue, err-count:16

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G01X81.500Y-61.500A-90.000'  line-Nr:40
Try to continue, err-count:17

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G02X61.500Y-81.500I-20.000J0.000A-180.000'  line-Nr:41
Progress: 85%  line:    41   time:22:53:12
Try to continue, err-count:18

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G01X18.500Y-81.500A-180.000'  line-Nr:42
Try to continue, err-count:19

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
< error:20 	Unsupported or invalid g-code command found in block.
! Last processed 'G02X-1.500Y-61.500I0.000J20.000A-270.000'  line-Nr:43
Try to continue, err-count:20
[MSG:Pgm End]
[Program end: 2025-02-02 22:53:16 ]

[Streaming finish line:23]

Accessory State run times :
             h: m: s
Spindle: 00000:59:19
---------------------------
< ok
< ok
-------------------------------------------------------------------------------------------------------

This is my setup for the rounded square

[image]

And here is the setup for the A-axis

[image]

svenhb, 2025-02-03:
What controller do you use and which grbl version is installed?
Does your grbl-controller supports an A-axis? An Arduino Uno/Nano with regular grbl does not (https://github.com/gnea/grbl/wiki#nice-features : "Grbl is for three axis machines. No rotation axes (yet) – just X, Y, and Z.").

Arduino Mega and other do support A-axis: https://github.com/fra589/grbl-Mega-5X

asoka777, 2025-02-03:
I'm using GRBL-Plotter and grbl-1.1h.20190825 and following the instructions from this link: https://grbl-plotter.de/index.php?id=tangential-knife. I'm wondering if using the A-axis as described in Example 1 allows me to use the Z-axis for lifting the knife.

I'm also considering using an Arduino Mega with RAMPS 1.4 and I'm not sure if GRBL-Plotter supports this setup.

svenhb, 2025-02-04:
Yes, there are two examples:
- Example 1: If you use a four axis controller, like Arduino Mega - then you can use 'A' for rotation and keep 'Z' for up/down.
- Example 2: if you use a three axis controller like Arduino Uno/Nano - then you need to use 'Z' for rotation and could use a Rc-servo (controlled via spindle-PWM) for up/down.

>I'm also considering using an Arduino Mega with RAMPS 1.4 and I'm not sure if GRBL-Plotter supports this setup.

It will support RAMPS, as long as you install a grbl firmware on it, like https://github.com/fra589/grbl-Mega-5X

asoka777, 2025-02-11:
Firstly, I apologize for the delayed response. I was busy testing the setup in every possible way based on your previous guidance. Unfortunately, I wasn’t successful and ended up burning some drivers and the board. I’ve now replaced the components and am ready to start fresh by following your instructions step by step.  

I need to use the **Z-axis** for up/down movement since I need the blade to penetrate PVC foam board, which I believe a servo won’t be able to handle. However, I still need help configuring **'A' for rotation** using an **Arduino Mega**.  

I have a question regarding the setup:  
1. If I go with **Example 1**, which setup should I use?  
   a) **Arduino Mega + RAMPS with GRBL-Mega-5X firmware**  
   b) **Arduino Mega + CNC Shield with standard GRBL firmware**  

I realize I need some serious guidance here, and I truly appreciate your patience. Sorry if I’m being a bit annoying, but I really hope you can help me get this project up and running.  

Looking forward to your advice!

svenhb, 2025-02-12:
Yes there are two options for the Mega, as I know:
the "regular" grbl: https://github.com/gnea/grbl-Mega, which seems to be a bit out dated (no code update since longer time) - but it also support just three axis?
and the "Mega-5X": https://github.com/fra589/grbl-Mega-5X, which has much more options in my eyes (additional digital in/out).

So I would go with the Mega-5X. I also plan to use it for my next project... Also the Ramps platform is cheap to get...

svenhb, 2025-04-12:
Hope it is solved

closed 2025-04-12 by svenhb