svenhb/GRBL-Plotter #436: SVG Element Filter

Stoli123, 2025-02-25:
Is there a way to exclude (or include) specific elements when doing an SVG to g-code conversion? For instance, I have an SVG file with PATH and TEXT elements, and I only want to include the PATH elements.

svenhb, 2025-02-25:
I don't know how many different elements you need to exclude...
Up to now an element filter is not implemented (but for color or stroke-width - check below).
You could import all, sort by geometry and delete figures manually:
[image]

Or exclude specific colors from import (application: https://youtu.be/ysZ6TaL_rZw?si=5_hefT859s4qjFRt&t=161):
[image]

Stoli123, 2025-02-25:
Thanks. I think sorting the code blocks by Geometry would be a solution, but for some reason the geometry info is not present after the conversion. Only the colors and widths seem to be present. Probably the format of the SVG in the source file. It's not worth spending much time on. I can easily modify the svg with a text editor to find and remove the elements before importing. Or, I bet if I ran it through Inkscape and resaved the file it might work. Thanks for the quick response.

svenhb, 2025-02-26:
>Probably the format of the SVG in the source file

Can you upload this file here, so I can check what's wrong?

Stoli123, 2025-02-26:
One of the files that I was playing with (this is rather large) can be found here:

[https://upload.wikimedia.org/wikipedia/commons/c/c2/Boston_Massachusetts_US_street_map.svg](url)

This is a smaller file that does the same:

[logan_svg.zip](https://github.com/user-attachments/files/18984097/logan_svg.zip)

I noticed they both throw the same warning :  
( Warning-2002: high amount of Figure objects. count:4654.  max:1000)
( Figure codes will be summarized to one Figure.)

Is that the problem?

Stoli123, 2025-02-26:
Thanks. That's why I thought it might be an easy solution to filter out all the <text> elements during the import. Since the text is also in there as <path> elements, it would just remove the duplication.

Stoli123, 2025-02-26:
Thanks. I have an easy solution for now. All the files I have been using that have this issue can be fixed with a quick find and replace in Notepad++ before I import into GRBL-Plotter. Replacing: ` (<text).*?(</text>)` with nothing has removed all the duplicate text. This works because ALL the text-elements have fill-opacity="0". It could be handy down the road to implement your idea since some future files may have visible text-elements as well.

svenhb, 2025-02-26:
>So fill-opacity="0" is something like a "hidden-layer", will try to take it into account here - but will take some time:

Check implementation in new version: https://github.com/svenhb/GRBL-Plotter/releases/tag/v.1.7.5.2

Stoli123, 2025-02-27:
That worked great. Thanks Sven. Turns out that there were a few text elements in the logan.svg file that did not have a fill-opacity=0, so your fix is a better solution than the find/replace option that I was using.

svenhb, 2025-02-28:
Great, thanks for the feedback.