svenhb/GRBL-Plotter #438: Time Estimates

Stoli123, 2025-03-17:
I was wondering if you have ever explored the area of plot time estimation while taking the plotter's acceleration, deceleration and vector changes into account. I have been creating some lengthy plots using Drawbot V2 code and my time estimates are way off, presumably because of all those variables add up over a long plot. We know the speeds in the g-code set by F, and we know the specific machine's accel and max travel rate settings (my plotter repositions in xy at 9000mm/min and accel is 200mm/sec^2, z axis is 1200mm/min and 80mm/sec^2), but I feel that the change in the pen's vector from point to point is having a significant impact on time. Would you agree? Do you know how GRBL calculates pen speed based on those directional changes?

svenhb, 2025-03-18:
>I was wondering if you have ever explored the area of plot time estimation while taking the plotter's acceleration, deceleration and vector changes into account.

I do the calculations twice - during graphics import and when analyzing G-Code...
During graphics-import:
To be hornest I do it the simplest way: I add all pen-down path and finally I divide the distance by the XY-Feedrate. Additíonally I add the times for Z-movement and G04 delay.

Analyzing G-Code:
Depending on which axes are moving, the slowest feed rate will be taken. For rapid I assume 5000 (as far as I can see right now). Acceleration is not taken into account. And on the first sight also G04 is not taken into account.

>but I feel that the change in the pen's vector from point to point is having a significant impact on time. Would you agree? 

Yes, as I know the slowest feedrate of all movinga axes will be taken and also the max. accelaration for each axis will never exceeded.

svenhb, 2025-04-12:
I tried to take acceleration into account, but the resulting time for large G-Code is still off....

closed 2025-04-12 by svenhb

Stoli123, 2025-04-12:
Same here. I tried to implement an algorithm to take the acceleration into account, but my results varied quite a bit and eventually I moved on to other things.