svenhb/GRBL-Plotter #439: Add "tick box": "rotate around 0;0" for "Rotate by ...°" at G-Code Transform

oscer959, 2025-04-01:
If using GRBL-Plotter for G-code transformation it would be nice to get an option to easily rotate the code around 0;0 (or any coordinate (if possible)). The functionality already exist in GRBL-Plotter, but only at "Machine control - Show Camera - Manual mode". It would be super convenient to have that "tick box" for Rotate around 0;0 also at "G-Code Transform - Rotate by ...°". It will save time when modifying nc/cnc-files.

If easier an other option would be to add possibility to specify the Angle directly without the "right click and release"-procedure within the "Machine control - Show Camera - Manual mode".

Finally I must thank for a good software. Regarding the rotate functionality GRBL-Plotter actually handles it equally good as InPlot CNC Editor (and even better than CIMCO Edit that struggles with rotation of some of my arcs). I also add a screenshot from InPlot rotate function. As a bonus it was also be nice if GRBL-Plotter can add the possibility to "copy and rotate ", e.g. create pattern around 0;0 with multiple code instances... But just a bonus, rotate single instance around 0;0 is the thing I need for now. https://i-logic.com/inplot/ 

[image]

svenhb, 2025-04-02:
Good idea, will try to implement soon.

svenhb, 2025-04-12:
Fixed with latest release: https://github.com/svenhb/GRBL-Plotter/releases

closed 2025-04-12 by svenhb

oscer959, 2025-04-14:
> Fixed with latest release: https://github.com/svenhb/GRBL-Plotter/releases

Thanks! Working good. :)